microsoft/AirSim #3691: MacOS - Build and Run issues during the installation

Dives44, 2021-05-13:
## Question
How to solve the error: Command /Users/Shared/Epic Games/UE_4.25/Engine/Build/BatchFiles/Mac/XcodeBuild.sh failed with exit code 6 while trying to build AirSim on MacOS ?

## Context
I want to use AirSim on MacOS so I followed these [instructions](https://microsoft.github.io/AirSim/build_linux/). However at this [step](https://microsoft.github.io/AirSim/build_linux/#mac) "_Open the XCode workspace, and press the Build and run button in the top left_" I obtain the error mentioned above.

### Details
I'm very to Xcode and Unreal Engine so I do not really know what piece of information you need. However Xcode also give this message:

`The Legacy Build System will be removed in a future release. You can configure the selected build system and this deprecation message in File > Workspace Settings.`

Here is the entire error:
`Running bundled mono, version: Mono JIT compiler version 5.16.0.220 (2018-06/bb3ae37d71a Fri Nov 16 17:12:11 EST 2018)
Building BlocksEditor...
2021-05-13 14:07:31.285 defaults[11780:59313] 
The domain/default pair of (com.apple.dt.Xcode, IDEBuildOperationMaxNumberOfConcurrentCompileTasks) does not exist
Building ShaderCompileWorker...
Creating makefile for ShaderCompileWorker (no existing makefile)
ERROR: Couldn't find target rules file for target 'ShaderCompileWorker' in rules assembly 'UE4Rules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.
       Location: /Users/Shared/Epic Games/UE_4.25/Engine/Intermediate/Build/BuildRules/UE4Rules.dll
       Target rules found:
Running command : Engine/Binaries/DotNET/UnrealBuildTool.exe BlocksEditor Mac Development /Users/julien/AirSim/Unreal/Environments/Blocks/Blocks.uproject -buildscw
Creating makefile for BlocksEditor (no existing makefile)
ERROR: Unable to find plugin 'AirSim' (referenced via Blocks.uproject). Install it and try again, or remove it from the required plugin list.
Command /Users/Shared/Epic Games/UE_4.25/Engine/Build/BatchFiles/Mac/XcodeBuild.sh failed with exit code 6`

### What I found
If I switch from the Legacy Build System to the New Build System I obtain a different error:
`Running bundled mono, version: Mono JIT compiler version 5.16.0.220 (2018-06/bb3ae37d71a Fri Nov 16 17:12:11 EST 2018)
Building BlocksEditor...
2021-05-13 14:24:50.223 defaults[19619:76481] 
The domain/default pair of (com.apple.dt.Xcode, IDEBuildOperationMaxNumberOfConcurrentCompileTasks) does not exist
Building ShaderCompileWorker...
Creating makefile for ShaderCompileWorker (no existing makefile)
ERROR: Couldn't find target rules file for target 'ShaderCompileWorker' in rules assembly 'UE4Rules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.
       Location: /Users/Shared/Epic Games/UE_4.25/Engine/Intermediate/Build/BuildRules/UE4Rules.dll
       Target rules found:
Running command : Engine/Binaries/DotNET/UnrealBuildTool.exe BlocksEditor Mac Development /Users/julien/AirSim/Unreal/Environments/Blocks/Blocks.uproject -buildscw
Creating makefile for BlocksEditor (no existing makefile)
ERROR: Unable to find plugin 'AirSim' (referenced via Blocks.uproject). Install it and try again, or remove it from the required plugin list.
Command ExternalBuildToolExecution failed with a nonzero exit `code``

## System
- MBP Intel on MacOS BigSur 11.2.3
- Unreal Engine 4.25.4
- xcode-select version 2384

Can someone help me plz 🙏

rajat2004, 2021-05-13:
You could try the steps mentioned in https://answers.unrealengine.com/questions/354265/ue4editor-error-no-shadercompileworker-binary.html?sort=oldest, but the error seems to be strange.
The missing AirSim plugin seems to indicate the previous steps of building AirSim might not have completed correctly, could you post the output of `setup.sh` and `build.sh`

Dives44, 2021-05-13:
Thanks rajat you're right, I have no error with `./setup.sh` (it says "_AirSim setup completed successfully!_") but there is in fact an error when running `./build.sh`:

```
+ debug=false
+ [[ 0 -gt 0 ]]
+ '[' '!' -d ./external/rpclib/rpclib-2.2.1 ']'
+ '[' -d ./cmake_build ']'
++ which cmake
+ CMAKE=/opt/local/bin/cmake
+ false
+ build_dir=build_release
++ uname
+ '[' Darwin == Darwin ']'
+ export CC=/usr/local/opt/llvm@8/bin/clang
+ CC=/usr/local/opt/llvm@8/bin/clang
+ export CXX=/usr/local/opt/llvm@8/bin/clang++
+ CXX=/usr/local/opt/llvm@8/bin/clang++
+ [[ -d ./AirLib/deps/eigen3/Eigen ]]
+ echo 'putting build in build_release folder, to clean, just delete the directory...'
putting build in build_release folder, to clean, just delete the directory...
+ [[ -f ./cmake/CMakeCache.txt ]]
+ [[ -d ./cmake/CMakeFiles ]]
+ folder_name=
+ [[ ! -d build_release ]]
+ mkdir -p build_release
+ pushd build_release
+ false
+ folder_name=Release
+ /opt/local/bin/cmake ../cmake -DCMAKE_BUILD_TYPE=Release
-- The C compiler identification is Clang 8.0.1
-- The CXX compiler identification is Clang 8.0.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /usr/local/opt/llvm@8/bin/clang
-- Check for working C compiler: /usr/local/opt/llvm@8/bin/clang - broken
CMake Error at /opt/local/share/cmake-3.19/Modules/CMakeTestCCompiler.cmake:66 (message):
  The C compiler

    "/usr/local/opt/llvm@8/bin/clang"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/julien/Airsim/build_release/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make cmTC_d3c84/fast && /Applications/Xcode.app/Contents/Developer/usr/bin/make  -f CMakeFiles/cmTC_d3c84.dir/build.make CMakeFiles/cmTC_d3c84.dir/build
    Building C object CMakeFiles/cmTC_d3c84.dir/testCCompiler.c.o
    /usr/local/opt/llvm@8/bin/clang   -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk -mmacosx-version-min=11.2 -o CMakeFiles/cmTC_d3c84.dir/testCCompiler.c.o -c /Users/julien/Airsim/build_release/CMakeFiles/CMakeTmp/testCCompiler.c
    clang-8: error: invalid version number in '-mmacosx-version-min=11.2'
    make[1]: *** [CMakeFiles/cmTC_d3c84.dir/testCCompiler.c.o] Error 1
    make: *** [cmTC_d3c84/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/julien/Airsim/build_release/CMakeFiles/CMakeOutput.log".
See also "/Users/julien/Airsim/build_release/CMakeFiles/CMakeError.log".
+ popd
~/Airsim ~/Airsim
+ rm -r build_release
+ exit 1

```

rajat2004, 2021-05-14:
It might be that clang-8 is too old for OSX 11.2, you could try using a newer clang and see if it fixes the problem. I don't have a Mac so can't test personally. Some mixup of XCode and other tools is also possible, but less likely.

Dives44, 2021-05-14:
Ok I see, how can I tell the program to use a newer clang version ?
I already have 12.0.5 as default apparently:
```
$clang --version
Apple clang version 12.0.5 (clang-1205.0.22.9)
Target: x86_64-apple-darwin20.3.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

rajat2004, 2021-05-14:
Just change the below lines to `clang` and `clang++` instead of the absolute path -

https://github.com/microsoft/AirSim/blob/fa87bfe2a26b272a4cde1d0060a1eb30a1a01887/build.sh#L52-L54

Dives44, 2021-05-15:
Yes It works ! 

However I still have an Xcode issue.. Here is my new error:
`Running bundled mono, version: Mono JIT compiler version 5.16.0.220 (2018-06/bb3ae37d71a Fri Nov 16 17:12:11 EST 2018)
Building BlocksEditor...
2021-05-15 11:38:33.125 defaults[5505:71921] 
The domain/default pair of (com.apple.dt.Xcode, IDEBuildOperationMaxNumberOfConcurrentCompileTasks) does not exist
Building ShaderCompileWorker...
Creating makefile for ShaderCompileWorker (no existing makefile)
ERROR: Couldn't find target rules file for target 'ShaderCompileWorker' in rules assembly 'UE4Rules, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null'.
       Location: /Users/Shared/Epic Games/UE_4.25/Engine/Intermediate/Build/BuildRules/UE4Rules.dll
       Target rules found:
Running command : Engine/Binaries/DotNET/UnrealBuildTool.exe BlocksEditor Mac Development /Users/julien/AirSim/Unreal/Environments/Blocks/Blocks.uproject -buildscw
Creating makefile for BlocksEditor (no existing makefile)
WARNING: Library '/System/Library/PrivateFrameworks/MultitouchSupport.framework/Versions/Current/MultitouchSupport' was not resolvable to a file when used in Module 'Core', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
WARNING: Library '/System/Library/PrivateFrameworks/MultitouchSupport.framework/Versions/Current/MultitouchSupport' was not resolvable to a file when used in Module 'ApplicationCore', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
Parsing headers for BlocksEditor
  Running UnrealHeaderTool "/Users/julien/AirSim/Unreal/Environments/Blocks/Blocks.uproject" "/Users/julien/AirSim/Unreal/Environments/Blocks/Intermediate/Build/Mac/BlocksEditor/Development/BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -abslog="/Users/julien/Library/Application Support/Epic/UnrealBuildTool/Log_UHT.txt" -installed
Reflection code generated for BlocksEditor in 8,5224567 seconds
Building BlocksEditor...
Performing 7 actions (4 in parallel)
[1/7] Compile SharedPCH.Engine.ShadowErrors.h
[2/7] Compile Module.AirSim.cpp
[3/7] Compile Module.AirSim.gen.cpp
[4/7] Compile Blocks.cpp
[5/7] Link UE4Editor-AirSim.dylib
clang: error: no such file or directory: '/System/Library/PrivateFrameworks/MultitouchSupport.framework/Versions/Current/MultitouchSupport'
Command /Users/Shared/Epic Games/UE_4.25/Engine/Build/BatchFiles/Mac/XcodeBuild.sh failed with exit code 6`

Apparently its about multitouch or smith related. I found [this](https://forums.unrealengine.com/t/ue4-big-sur-xcode/157519) information but it is about UE 4.24 and I'm using UE 4.25 as recommended. The solution imply dealing with inner AirSim or Unreal code and I'm not sure it is the right path..

Dives44, 2021-05-15:
According to this [page](https://community.gamedev.tv/t/cannot-create-third-person-project-error/157020) the location of multitouch have changed in BigSure and it has been taken into account only in UE 4.26. Consequently I change my UE version and it seems to work. At least the main problem of this thread has been solved. 

Thanks a lot @rajat2004 🙏

rajat2004, 2021-05-15:
Glad to know that the problem has been fixed. You can just close the issue

The documentation and script needs to be updated though

PR #4370 (2022-02-25): Enable AirSim build on MacOS Big Sur

Fixes: #3691    

## About
This enables the AirSim build to run on MacOS Big Sur 11.5 by changing from LLVM@8 to LLVM@12. This is a continuation of #4021 with code review feedback applied.

## How Has This Been Tested?
MacOS build confirmed to run.

## Screenshots (if appropriate):
closed 2021-05-16 by Dives44