microsoft/AirSim #3696: Compile on M1 Mac Big Sur w/ Unreal Engine 4.25.4

christosachilleoudis, 2021-05-14:
Hi everyone! I have recently been trying to build AirSim on my M1 Mac. I read through issue #1898 and saw that there is a solution for Mac OS X with Unity created by @klintan. 

Here are my currents specs: 
laptop: MacBook Air M1 Chip
OS: macOS Big Sur (11.3.1)
Xcode: downloaded from the App Store (Version 12.5) 

I installed Unreal Engine 4.25.4 via the following article:
https://medium.com/techiepedia/setting-up-unreal-engine-m1-macbook-for-c-game-development-eb40c12237d1

I then cloned AirSim, successfully ran the setup.sh file with no problems. 

But when I run the build.sh file I get lots tons of errors. I am not really sure where to start to fix this. 

Any help or insight would be much appreciated! This is also my first ever post on Github Issues (I'm a total noob) so please let me know if there is any more info that would be helpful. 

Thanks!

Here is my terminal output:

Last login: Thu May 13 15:59:11 on ttys000
christos@MacBook-Air ~ % cd Documents/Code 
christos@MacBook-Air Code % ls
AirSim
christos@MacBook-Air Code % cd AirSim 
christos@MacBook-Air AirSim % ls
AUTHORS.md				UnrealPluginFiles.vcxproj
AirLib					UnrealPluginFiles.vcxproj.filters
AirLibUnitTests				azure
AirSim.props				build.cmd
AirSim.sln				build.sh
AirSim.sln.vlconfig			build_docs.bat
CHANGELOG.md				build_docs.sh
CONTRIBUTING.md				build_release
DroneServer				check_cmake.bat
DroneShell				clean.cmd
Examples				clean.sh
HelloCar				clean_rebuild.bat
HelloDrone				clean_rebuild.sh
HelloSpawnedDrones			cmake
LICENSE					docker
LogViewer				docs
MavLinkCom				external
PX4Scripts				install_run_all.sh
PythonClient				install_unreal.sh
README.md				mkdocs.yml
SGM					pipelines
SUPPORT.md				ros
Unity					setup.sh
Unreal					tools
christos@MacBook-Air AirSim % ./setup.sh
+ set -e
+++ dirname ./setup.sh
++ cd .
++ pwd
+ SCRIPT_DIR=/Users/christos/Documents/Code/AirSim
+ pushd /Users/christos/Documents/Code/AirSim
+ downloadHighPolySuv=true
+ MIN_CMAKE_VERSION=3.10.0
+ [[ 0 -gt 0 ]]
++ uname
+ '[' Darwin == Darwin ']'
+ brew update
Updated 1 tap (homebrew/core).
==> Updated Formulae
libidn2 ✔           haskell-stack       pushpin             semgrep
aravis              percona-toolkit     sbtenv              shibboleth-sp
earthly             postgresql@9.6      scalaenv

You have 1 outdated formula installed.
[... 1341 lines cut ...]
/usr/local/opt/llvm@8/bin/../include/c++/v1/cstdlib:115:9: error: no member named 'strtol' in the global namespace
using ::strtol;
      ~~^
/usr/local/opt/llvm@8/bin/../include/c++/v1/cstdlib:117:9: error: no member named 'strtoll' in the global namespace
using ::strtoll;
      ~~^
/usr/local/opt/llvm@8/bin/../include/c++/v1/cstdlib:119:9: error: no member named 'strtoul' in the global namespace
using ::strtoul;
      ~~^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/dispatcher.cc.o] Error 1
20 errors generated.
make[2]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/this_handler.cc.o] Error 1
20 errors generated.
make[2]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/rpc_error.cc.o] Error 1
20 errors generated.
make[2]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/server.cc.o] Error 1
20 errors generated.
make[2]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/client.cc.o] Error 1
20 errors generated.
make[2]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/detail/server_session.cc.o] Error 1
make[1]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/all] Error 2
make: *** [all] Error 2
christos@MacBook-Air AirSim %

rajat2004, 2021-05-15:
Might be the same issue as https://github.com/microsoft/AirSim/issues/3691, try running `clean.sh` and then `build.sh` again with the suggested fix. Also, please put command output or scripts inside code fences, prevents markdown formatting it and makes it easier to read

christosachilleoudis, 2021-05-17:
Thank you @rajat2004 this fixed my problem! I really appreciate it!

closed 2021-05-17 by christosachilleoudis