microsoft/AirSim #3698: AirSim crashes as soon as it connects to PX4

vxj9800, 2021-05-16:
## Question
### What's your question?
AirSim crashes when I run it right after I initialize PX4. I am not sure what may cause this issue as I am new to AirSim.

### Include context on what you are trying to achieve
I want to create some MAVSDK based code. So, I decided to use PX4 in SITL mode with AirSim as the simulator. I first downloaded the pre-compiled binary with Blocks environment. Everything works fine when I run the Blocks.sh without changing settings.json file in the Documents/AirSim/ folder. I was able to run both car and quadcopter model without any issues. Then I followed all the steps provided in https://microsoft.github.io/AirSim/px4_sitl/ to install and setup PX4 to work with AirSim. However, AirSim crashes as soon as I get the log `INFO  [simulator] Simulator connected on TCP port 4560.` on PX4. The complete AirSim logs for both the cases are provided below.

**AirSim log without changing settings.json**
[logAirSim.txt](https://github.com/microsoft/AirSim/files/6489418/logAirSim.txt)

**AirSim log with changed settings.json**
[logAirSimPx4.txt](https://github.com/microsoft/AirSim/files/6489428/logAirSimPx4.txt)
I also get following message in terminal
```
terminating with uncaught exception of type std::runtime_error: UdpClientPort socket send failed with error: 9

Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Malloc Size=131160 LargeMemoryPoolOffset=459112 
Malloc Size=81006 LargeMemoryPoolOffset=540142 
Aborted (core dumped)
```

#### Context details
**OS**: Ubuntu 20.04
**AirSim**: Blocks v1.4.0
**PX4**:  v1.11.3
**settings.json**:
```
{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "ClockType": "SteppableClock",
    "Vehicles": {
        "PX4": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "LockStep": true,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPortLocal": 14540,
            "ControlPortRemote": 14580,
            "Sensors":{
                "Barometer":{
                    "SensorType": 1,
                    "Enabled": true,
                    "PressureFactorSigma": 0.0001825
                }
            },
            "Parameters": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "COM_OBL_ACT": 1,
                "LPE_LAT": 47.641468,
                "LPE_LON": -122.140165
            }
        }
    }
}
```

### Include details of what you already did to find answers
I also tried running PX4 with jMAVSim and it works fine. So, it seems that the issue is with AirSim. However, I don't know much about unreal engine and so I would prefer not to look at the source code of AirSim.

chungggg, 2021-05-17:
You can try with remove all element inside Sensors object in settings.json. I have same problem, when I remove all element inside Sensors object then AirSim connected to PX4 and QGroundControl. But I trying to fix with sensor because I need sensors for some task

chungggg, 2021-05-17:
I found problem by settings.json file. We need 3 sensors to resolve problem: IMU, Barometer and Magnetometer

vxj9800, 2021-05-18:
@chungggg With your suggestion, I used following settings.json file.
```
{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "ClockType": "SteppableClock",
    "Vehicles": {
        "PX4": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "LockStep": true,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPortLocal": 14540,
            "ControlPortRemote": 14580,
            "Sensors":{
                "Barometer": {
					"SensorType": 1,
					"Enabled" : true,
					"PressureFactorSigma": 0.0001825,
					"PressureFactorTau": 3600,
					"UncorrelatedNoiseSigma": 2.7,
					"UpdateLatency": 0,
					"UpdateFrequency": 50,
					"StartupDelay": 0

				},
				"Imu": {
					"SensorType": 2,
					"Enabled" : true,
					"AngularRandomWalk": 0.3,
					"GyroBiasStabilityTau": 500,
					"GyroBiasStability": 4.6,
					"VelocityRandomWalk": 0.24,
					"AccelBiasStabilityTau": 800,
					"AccelBiasStability": 36
				},
				"Gps": {
					"SensorType": 3,
					"Enabled" : true,
					"EphTimeConstant": 0.9,
					"EpvTimeConstant": 0.9,
					"EphInitial": 25,
					"EpvInitial": 25,
					"EphFinal": 0.1,
					"EpvFinal": 0.1,
					"EphMin3d": 3,
					"EphMin2d": 4,
					"UpdateLatency": 0,
					"UpdateFrequency": 50,
					"StartupDelay": 0
				},
				"Magnetometer": {
					"SensorType": 4,
					"Enabled" : true,
					"NoiseSigma": 0.005,
					"ScaleFactor": 1,
					"NoiseBias": 0,
					"UpdateLatency": 0,
					"UpdateFrequency": 50,
					"StartupDelay": 0
				}
            },
            "Parameters": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "COM_OBL_ACT": 1
            }
        }
    }
}
```
AirSim doesn't crash with this file. However, I get following output in PX4.
```
ERROR [simulator] poll timeout 0, 111
ERROR [simulator] poll timeout 0, 111
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2021-05-18/01_38_35.ulg
INFO  [logger] Opened full log file: ./log/2021-05-18/01_38_35.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [mavlink] partner IP: 127.0.0.1
ERROR [ekf2] start failed (-1)
```

It is clear that PX4 is not able to get the ekf to work. Can you please tell me what I might have wrong in the settings.json file?

jonyMarino, 2021-06-01:
Hi @vxj9800! How was it?

vxj9800, 2021-06-02:
Hello @jonyMarino. Sorry for the delay in response. The new versions worked fine for the most part. However, I did get following errors popping up every now and then.
```
[2021.06.02-00.37.18:284][  0]LogStreaming: Error: Couldn't find file for package /AirSim/Weather/WeatherFX/WeatherActor requested by async loading code. NameToLoad: /AirSim/Weather/WeatherFX/WeatherActor
[2021.06.02-00.37.18:284][  0]LogStreaming: Error: Found 0 dependent packages...
[2021.06.02-00.37.18:284][  0]LogTemp: Warning: Warning, WeatherAPI got invalid weather actor class!
[2021.06.02-00.37.18:284][  0]LogStreaming: Error: Couldn't find file for package /AirSim/Weather/UI/MenuActor requested by async loading code. NameToLoad: /AirSim/Weather/UI/MenuActor
[2021.06.02-00.37.18:284][  0]LogStreaming: Error: Found 0 dependent packages...
[2021.06.02-00.37.18:284][  0]LogTemp: Warning: Warning, WeatherAPI got invalid menu actor class!
```
I don't know how critical these error are but they didn't crash the AirSim. Also, the AirSim hangs when the PX4 is shutdown before closing the AirSim window.

In any case, the latest version is stable enough for me and solves my issue. Thank you very much!

jonyMarino, 2021-06-02:
That error is not related to this and is not relevant if you are not using the weather options. I already saw it, and we fixed it. I don't remember where ... which environment are you using?

vxj9800, 2021-06-02:
I am using 'Blocks' environment. Haven't tried any other environment yet. I will give the other environments a shot and let you know if this error persists.

lovettchris, 2021-06-04:
Can you check this PR fixes it?  
https://github.com/microsoft/AirSim/pull/3757

closed 2021-06-18 by zimmy87

zimmy87, 2021-06-21:
For the issue mentioned earlier in this thread about the hang when PX4 is shutdown before AirSim, I've filed a new issue #3807 to track conversation for that particular issue separately. The original issue described here looks like its been fixed for some time.

PR #3757 (2021-06-18): Fix bug in UDP socket failing with WSAECONNRESET in the read thread

Fixes: #3698    
Fixes: #2924

## About
Fix bug in UDP socket failing with WSAECONNRESET in the read thread if it tries to read to soon.  The fix is to put a retry count in the UdpClientPort where it automatically reconnects up to 10 times.  Also fixed a null ref crash on mav_vehicle_ that happens in some cases like if you kill px4 process while drone is flying.

## How Has This Been Tested?
Windows with PX4 in WSL2 and in WSL1 with latest PX4 master bits as of today (1.12.0)

## Screenshots (if appropriate):