microsoft/AirSim #3704: Nav mesh breaks when `AirSimGameMode` is started

blakermchale, 2021-05-19:
## Bug report 

- AirSim Version/#commit: aba8ff15a918b766d10dccc93dba60b91b256635
- UE/Unity version: 4.26
- autopilot version: PX4
- OS Version: Windows 10

### What's the issue you encountered?

When `AirSimGameMode` is started the nav meshes collapse into a few stacked squares. If I start it without AirSim everything works fine. This is an issue since some AI I am writing cannot navigate without a nav mesh.

`AirSimGameMode` running:
[image]

Default `GameMode` running:
[image]

### Settings

Using default.

### How can the issue be reproduced?

1. Start Blocks environment in UE4 4.26
2. Add `Nav Mesh Bounds Volume` to world and scale to large size
3. Run `AirSimGameMode`
4. Press `F8` to eject and `p` to view nav meshes

### Include full error message in text form
N/A

**What's better than filing an issue? Filing a pull request :).**

blakermchale, 2021-05-19:
I found a temporary fix to the problem. If you go to `Project Settings > Navigation Mesh > Runtime > Runtime Generation` and set it to `Dynamic` instead of `Static` it will properly load the nav mesh.

emersonjr, 2021-05-26:
thank you very much @blakermchale  i was having basically the same problem, it was blocking my pedestrian system to work.

for learning reasons, how did you figure out that it would solve the problem? how did you spot it?

blakermchale, 2021-05-26:
I was looking up problems with the navmesh disappearing in UE4 and stumbled across [this stackoverflow](https://answers.unrealengine.com/questions/83075/nav-mesh-problemsdisappearing.html). It wasn't a solution to this problem, but it pointed me to that settings area. I knew there was a problem where the mesh had its position corrupted, so I tried changing the setting from Static to Dynamic in hopes that it would reload the mesh correctly and it ended up doing it.

jonyMarino, 2021-06-14:
Hi @blakermchale! Thanks for keeping the issue updated. To diagnose what is happening. If you put the player start in position 0,0,0 before starting AirSim. Do you also have this issue?

blakermchale, 2021-06-15:
@jonyMarino  I put the players start position to 0,0,0 and it fixed the issue!

jonyMarino, 2021-06-15:
Well. The center of the unreal world is shifted to the player's starting position at the beginning. Those meshes are seen to be affected by that. @zimmy87, can you think of a way to tell those meshes not to change their position with that of the unreal environment?

linchang77, 2025-04-13:
wow,thank you very much,This problem has been distressing me for three days, now i can push the project forward again