microsoft/AirSim #3706: CityEnviron - false collision detected

andreasceid, 2021-05-19:
-----------------
## Bug report 
-----------------

- AirSim Version: 1.4
- UE version: 4.26.1
- autopilot version: N/A
- OS Version: Microsoft Windows 10 Education Version 10.0.19042 Build 19042

-----------------
### What's the issue you encountered?
-----------------

I'm currently training a reinforcement learning agent using AirSim in `CityEnviron` environment. The drone follows a specific path in the beginning (because I don't use epsilon greedy method). On that path, the **collision API** reports a collision. The weird part is that it doesn't apply with the visual input that I have (I am constantly monitoring the `CityEnviron` window). I'll attach several paths that lead to the same endpoint approximately for test case:

- Path 1:
    ```py
    [  0,   0,   0] - Collision: False
    [ -5,   0, -20] - Collision: False
    [-10,   0, -19] - Collision: False
    [-16,   0, -19] - Collision: False
    [-21,   0, -19] - Collision: False
    [-26,   0, -19] - Collision: False
    [-31,   0, -18] - Collision: False
    [-36,   0, -18] - Collision: False
    [-39,   0, -18] - Collision: True
    ```
- Path 2:
    ```py
    [  0,   0,   0] - Collision: False
    [ -5,   0, -20] - Collision: False
    [-10,   0, -20] - Collision: False
    [-15,   0, -19] - Collision: False
    [-20,   0, -19] - Collision: False
    [-26,   0, -19] - Collision: False
    [-31,   0, -18] - Collision: False
    [-36,   0, -18] - Collision: False
    [-41,   0, -18] - Collision: True
    ```
- Path 3:
    ```py
    [  0,   0,   0] - Collision: False
    [ -5,   0, -20] - Collision: False
    [-10,   0, -19] - Collision: False
    [-15,   0, -19] - Collision: False
    [-20,   0, -19] - Collision: False
    [-26,   0, -18] - Collision: False
    [-31,   0, -18] - Collision: False
    [-36,   0, -18] - Collision: False
    [-41,   0, -18] - Collision: True
    ```
- Path 4:
    ```py
    [  0,   0,   0] - Collision: False
    [ -5,   0, -20] - Collision: False
    [-10,   0, -19] - Collision: False
    [-15,   0, -19] - Collision: False
    [-20,   0, -19] - Collision: False
    [-26,   0, -18] - Collision: False
    [-31,   0, -18] - Collision: False
    [-36,   0, -18] - Collision: False
    [-42,   0, -18] - Collision: True
    ```
- Path 5:
    ```py
    [  0,   0,   0] - Collision: False
    [ -5,   0, -20] - Collision: False
    [-10,   0, -19] - Collision: False
    [-15,   0, -19] - Collision: False
    [-20,   0, -19] - Collision: False
    [-26,   0, -18] - Collision: False
    [-31,   0, -18] - Collision: False
    [-37,   0, -18] - Collision: False
    [-42,   0, -18] - Collision: True
    ```

-----------------
### Settings
-----------------

```json
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ClockType": "SteppableClock",
  "ClockSpeed": 1000.0,
  "ViewMode": "FlyWithMe" 
}
```
-----------------
### How can the issue be reproduced?
-----------------

I'll attach a sample code to indicate the method I use to move the drone. The actual code is much bigger, and in experimental mode, but I think you'll be able to find the bug given the description above:

```py
import time
import numpy
import setup_path
import airsim

def interpret_action(action, v_decay_counter):
    if action == 0:
        quad_offset = (5, 0, 0)
    elif action == 1:
        quad_offset = (0, 5, 0)
    elif action == 2:
        quad_offset = (0, 0, 5)
    elif action == 3:
        quad_offset = (-5, 0, 0)
    elif action == 4:
        quad_offset = (0, -5, 0)
    else:
        quad_offset = (0, 0, -5)

    if v_decay_counter >= 4:
        v_decay_counter = 0
    for i in range(v_decay_counter):
        quad_offset = tuple([momentum*x for x in quad_offset])
    v_decay_counter += 1
    return v_decay_counter

v_decay_counter = 0
momentum = 0.6
clockspeed = 1000

client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)
client.moveToZAsync(z=-20, velocity=10, timeout_sec=10).join()
collision = client.simGetCollisionInfo().has_collided

while collision is False:

    quad_offset, v_decay_counter = interpret_action(3, v_decay_counter)

    client.moveByVelocityAsync(
        quad_offset[0],
        quad_offset[1],
        quad_offset[2],
        2,
        drivetrain=airsim.DrivetrainType.ForwardOnly,
        yaw_mode=airsim.YawMode(False, 0)
    ).join()
    time.sleep(2 / clockspeed)
    client.moveByVelocityAsync(0, 0, 0, 1).join()
    client.hoverAsync().join()

    print("Position: %s" % client.kinematics_estimated.position)

    collision = client.simGetCollisionInfo().has_collided
```
-----------------
### Include full error message in text form

N/A

**What's better than filing an issue? Filing a pull request :).**

zimmy87, 2021-06-01:
Hi @AndreasKaratzas! Thank you for reporting this issue. The collision you're seeing is due to a building that is marked visible = false, but still has it's collision profile set to default. The environment has been updated to make this building visible. The updated environment can be downloaded [here](https://github.com/microsoft/AirSim/releases/tag/v1.5.0-windows).

AndreasKaratzas, 2021-06-01:
Perfect :) Is there an updated `CityEnviron` targeted for Linux users, too?

zimmy87, 2021-06-01:
Unfortunately, `CityEnviron` doesn't compile on linux, so it is currently a windows-only environment.

closed 2021-06-02 by AndreasKaratzas