microsoft/AirSim #3707: Unexpected client behaviour - doesn't follow python instructions

andreasceid, 2021-05-19:
## Question

<br>

-----------
### What's your question?
-----------

I'm trying to execute a piece of python code, and the `Multirotor` client won't even take off. Is there something wrong with my code?

```py
import os
import time
import numpy
import random
import setup_path
import airsim

def interpret_action(action, momentum, v_decay_counter):
    if action == 0:
        quad_offset = (2, 0, 0)
    elif action == 1:
        quad_offset = (0, 2, 0)
    elif action == 2:
        quad_offset = (0, 0, 2)
    elif action == 3:
        quad_offset = (-2, 0, 0)
    elif action == 4:
        quad_offset = (0, -2, 0)
    else:
        quad_offset = (0, 0, -2)

    if v_decay_counter >= 4:
        v_decay_counter = 0
    for i in range(v_decay_counter):
        quad_offset = tuple([momentum*x for x in quad_offset])
    v_decay_counter += 1
    return quad_offset, v_decay_counter

def do_action(client, momentum, v_decay_counter, clockspeed):
    quad_offset, v_decay_counter = interpret_action(
        random.randint(0, 5), 
        momentum, 
        v_decay_counter
    )
    
    client.moveByVelocityAsync(
        quad_offset[0],
        quad_offset[1],
        quad_offset[2],
        2,
        drivetrain=airsim.DrivetrainType.ForwardOnly,
        yaw_mode=airsim.YawMode(False, 0)
    ).join()
    time.sleep(2 / clockspeed)
    client.moveByVelocityAsync(0, 0, 0, 1).join()
    client.hoverAsync().join()

    print("Position: \n %s \n" % 
        client.
        getMultirotorState().
        kinematics_estimated.
        position.
        to_numpy_array().
        astype(int))

    return v_decay_counter

def main():
    v_decay_counter = 0
    momentum = 0.6
    clockspeed = 1000

    client = airsim.MultirotorClient()
    client.confirmConnection()
    client.enableApiControl(True)
    client.armDisarm(True)
    client.reset()
    client.simRunConsoleCommand("t.MaxFPS 5")
    
    client.moveToZAsync(z=-20, velocity=10, timeout_sec=10).join()
    time.sleep(10 / clockspeed)
    collision = client.simGetCollisionInfo().has_collided
    
    while not collision:
        v_decay_counter = do_action(client, momentum, v_decay_counter, clockspeed)
        collision = client.simGetCollisionInfo().has_collided
    
    client.armDisarm(False)
    client.reset()
    client.enableApiControl(False)

if __name__ == "__main__":
    main()
```
<br>

-----------
### Include context on what you are trying to achieve
-----------

The ultimate purpose is to instruct the drone by calling the `interpret_action` method in a higher `clockspeed` rate and lower `FPS`. I've overcome the inaccuracy of the `moveToPositionAsync` instruction by using `moveByVelocityAsync`. I would like to find a solution with minimum changes to the code attached above.

<br>

-----------

### Context details

-----------

AirSim version: 1.4

Python version: 3.8

UE: 4.26.1

OS: Windows 10

<br>

-----------
### Settings
-----------

```json
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ClockType": "SteppableClock",
  "ClockSpeed": 1000.0,
  "ViewMode": "FlyWithMe" 
}
```

rajat2004, 2021-05-19:
See [`reset()` API description](https://microsoft.github.io/AirSim/api_docs/html/#airsim.client.VehicleClient.reset), `enableApiControl()`, `armDisarm()` etc needs to called after this. Moving the `reset()` call before the 2 lines works for me

rajat2004, 2021-05-21:
Not sure, maybe try adding a small upward velocity component, generally in drones I think a small throttle up is needed, if the vehicle is not nicely tuned. Or you could try the `moveByVelocityZ` API and see if it makes any difference

andreasceid, 2021-05-21:
Got it. If you find anything out anything else, please comment :)

closed 2021-06-14 by jonyMarino