microsoft/AirSim #3716: I try to use ImageType.Depthplanar output depth image, but it outputs a gray scale image. Not depth image.

PeterParkerLogic, 2021-05-20:
## Bug report 

- AirSim Version/#commit:
- UE/Unity version:
- autopilot version:
- OS Version:

### What's the issue you encountered?

### Settings

### How can the issue be reproduced?

1.
2.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

skykim0609, 2021-05-21:
I had this same issue too.

saihv, 2021-06-20:
> This issue was there since the name of depth type has changed to depthplanar( >= 1.4.0) and not resolved yet.

Hi @skykim0609, do you happen to know the commit ID before which this issue did not exist? I just tested with the version of the code before the DepthPlanar change was made, and I still see this issue, although I was getting an RGB image, not a single channel image.

skykim0609, 2021-06-29:
> Hi @skykim0609, do you happen to know the commit ID before which this issue did not exist? I just tested with the version of the code before the DepthPlanar change was made, and I still see this [...]

According to my memory, until AirSim 1.3.0 I didn't have this issue...

MassimoClementi, 2021-07-20:
Hi, I am on AirSim 1.5.0 too and have the same issue of AirSim DepthPlanar acquisitions having pixel values which unfortunately seem not to follow what specified in the official documentation:

> For ImageType = DepthPlanar, you get depth in camera plane, i.e., all points that are plane-parallel to the camera have same depth.

The retrieved image seems to be instead a grayscale image with a single float channel.

For clarity, in the following section I report both the code I use to retrieve the image and the AirSim configurations I use.

API code:
```
    responses = client.simGetImages([
        airsim.ImageRequest("front-left-custom", airsim.ImageType.Scene, False, False),
        airsim.ImageRequest("front-right-custom", airsim.ImageType.Scene, False, False),
        airsim.ImageRequest("front-left-custom", airsim.ImageType.DepthPlanar, True, False)])

    frame_left_1d = np.fromstring(responses[0].image_data_uint8, dtype=np.uint8)
    frame_left = frame_left_1d.reshape(responses[0].height, responses[0].width, 3)

    frame_right_1d = np.fromstring(responses[1].image_data_uint8, dtype=np.uint8)
    frame_right = frame_right_1d.reshape(responses[1].height, responses[1].width, 3)

    depth_img_in_meters = airsim.list_to_2d_float_array(responses[2].image_data_float, 
                    responses[2].width, responses[2].height)
    depth_img_in_millimeters = depth_img_in_meters * 1000
    depth_mm_16bit = np.clip(depth_img_in_millimeters, 0, 65535)
```

AirSim settings:

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ClockSpeed": 1.0,
  "SubWindows": [
  {"WindowID": 0, "ImageType": 0, "CameraName": "front-left-custom", "Visible": true},
  {"WindowID": 2, "ImageType": 0, "CameraName": "front-right-custom", "Visible": true}
  ],
  "Vehicles": {
    "SimpleFlight": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      },
      "Cameras": {
        "front-left-custom": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 720,
              "Height": 1280,
              "FOV_Degrees": 60
            },
            {
              "ImageType": 1,
              "Width": 720,
              "Height": 1280,
              "FOV_Degrees": 60
            }
          ],
          "X": 0.50, "Y": -0.06, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        },
        "front-right-custom": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 720,
              "Height": 1280,
              "FOV_Degrees": 60
            }
          ],
          "X": 0.50, "Y": 0.06, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        }
      },
      "X": 0, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
  }
}
```

Any advance on the topic? Although DepthPerspective works fine, for my task I actually need the planar depth. Any possible workaround? Thanks in advance and have a good day.

closed 2021-08-09 by zimmy87