microsoft/AirSim #3719: getRotorState Error

KIYONG-KWON, 2021-05-20:
## Question
### What's your question?
// Setting.json
{
"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
"SettingsVersion": 1.5,
"SimMode": "Multirotor",

"Vehicles": {
"Drone1": {
"VehicleType": "SimpleFlight",
"X": 4,
"Y": 0,
"Z": -2,
"Yaw": -180
},
"Drone2": {
"VehicleType": "SimpleFlight",
"X": 8,
"Y": 0,
"Z": 0,
"Yaw": -180
}
}
}

// main.cpp
string droneID = "Drone1";
client.enableApiControl(true, droneID);
client.getRotorStates(droneID); // error on this line , rpc::rpc_error

///////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
Cound you advise me why error occurred at this line? other codes to get sensor datas such as gps, imu etc do not make error like this.

### Include context on what you are trying to achieve

#### Context details
I updated Airsim version as the lastest 1.5.0 and unreal version is 4.26.

### Include details of what you already did to find answers

rajat2004, 2021-05-21:
You're likely using the PyPi airsim package which needs to be updated, till then, try using the latest master Python library using the instructions in https://github.com/microsoft/AirSim/pull/3611

KIYONG-KWON, 2021-05-22:
Do I have to update python lib even if i am using c++ API?

rajat2004, 2021-05-22:
Ah no, sorry for missing that. I'll try to test on my system as well. Just to confirm, both the programs are running on the same system? And is the platform Windows or Linux?

KIYONG-KWON, 2021-05-22:
Both the programs are running on the same system and window 10 platform. Above code main.cpp is part of HelloDrone example  of Airsim Soluiton.

rajat2004, 2021-05-23:
So the `getRotorStates` API seems to be working fine for me, my additional code -
```cpp
auto rotor_states = client.getRotorStates();
std::cout << "Rotor States: " << rotor_states.timestamp << std::endl;
for (auto& state: rotor_states.rotors) {
    std::cout << state.thrust << " " << state.torque_scaler << " " << state.speed << std::endl;
}
```

I'm however getting exceptions with the sensor APIs if the sensors aren't specified, maybe a behavioural change from one of the recent PRs. And the sensor APIs are working fine for you but not the rotor state API?

KIYONG-KWON, 2021-05-25:
Based on your link #3726, i updated AIrSimSettings.hpp and rebuild
However, below error is not still solved. 
I think my problem is different with #3726 because other sensor signals come in well.
I changed Seeting.json file to default but it doesn't work as well.

> Exception raised by the API, something went wrong.
> rpclib: server could not find function 'getRotorStates' with argument count 1.

`//Setting.json
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor"
}`

` //main.cpp
auto rotor_states = client.getRotorStates();
        std::cout << "Rotor States: " << rotor_states.timestamp << std::endl;
        for (auto& state : rotor_states.rotors) {
            std::cout << state.thrust << " " << state.torque_scaler << " " << state.speed << std::endl;
        }`

Do you have any more doubts? Is there any possibility that an incomplete Python environment could affect the c++ API? Currently, Python is not properly active on my PC.

jonyMarino, 2021-05-25:
Hi and welcome @KIYONG-KWON! please run `pip install airsim --upgrade`
the new version was just released

KIYONG-KWON, 2021-05-28:
I solved the problem. 

I commanded **update_from_git.bat** on cmd window and rebuild Blocks.Sin(Airsim\Unreal\Environments\Blocks). 
After that, i can getRotorState form API. I think It was my basic mistake.

By the way, I'm not familiar with this open source environment, so can you explain how 'update_from_git.bat' solve this problem?

jonyMarino, 2021-06-01:
update_from_git.bat imports the plugin sources from the repository to your Unreal project. So the problem was that your UE project had an old plugin that didn't have the feature you were looking for; 'update_from_git.bat' updated the plugin (after you have done a git pull, of course)

closed 2021-06-04 by KIYONG-KWON