microsoft/AirSim #3734: What's the meaning of 6 kinematic quantities in getCarState API?

weigangwen, 2021-05-26:
## Question
### Is there any relationship between kinematics_estimated.linear_velocity and car_state.speed?What is the meaning of angular velocity?

### I want to get the speed  and the rotational speed in the coordinate system of the car rigid body itself,  such as Rigidbody.velocity and Rigidbody.angularVelocity

#### 

###

clovett, 2021-06-06:
According to [AirSim/PythonClient/airsim/typee.py](https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/types.py) you have CarState with it's `kinematics_estimated`:

```python
class CarState(MsgpackMixin):
    speed = 0.0
    gear = 0
    rpm = 0.0
    maxrpm = 0.0
    handbrake = False
    collision = CollisionInfo()
    kinematics_estimated = KinematicsState()
    timestamp = np.uint64(0)
```
and this is defined asS:

```python
class KinematicsState(MsgpackMixin):
    position = Vector3r()
    orientation = Quaternionr()
    linear_velocity = Vector3r()
    angular_velocity = Vector3r()
    linear_acceleration = Vector3r()
    angular_acceleration = Vector3r()
```

Where:
- `position` is x,y,z location, in local coordinates (0,0,0) is the home position where vehicle started.  
- `orientation` is a Quaterion describing rotation of drone
- `linear_velocity` is the current speed of the drone in x, y, z
- `angular_velocity` is the speed of rotation in each axis
- `linear_acceleration` is the acceleration of the drone in x, y, z
- `angular_acceleration` is the acceleration of the rotation of the drone in each axis

weigangwen, 2021-06-08:
@clovett thanks!

closed 2021-06-08 by weigangwen