microsoft/AirSim #3747: I want to use Airsim exe files for eyetracking with HTC vive, anybody could do that to help me?

Nilzfar, 2021-05-27:
## What feature are you suggesting?
### Overview:

### Smaller Details: 

### Nature of Request:  

- Addition
- Change
- Removal
  - 

## Why would this feature be useful?

clovett, 2021-06-06:
Presumably you plan to use the [HTC Vivie Eye Tracking SDK](https://developer.vive.com/resources/vive-sense/sdk/vive-eye-and-facial-tracking-sdk/), with it's unreal plugin and use the eye tracking info to do something in AirSim, what do you want the eyetracking to do, look around, or control the drone?

Nilzfar, 2021-06-07:
> Presumably you plan to use the [HTC Vivie Eye Tracking SDK](https://developer.vive.com/resources/vive-sense/sdk/vive-eye-and-facial-tracking-sdk/), with it's unreal plugin and use the eye tracking [...]

well, yes I want to just extract where my subject is looking in real-time in a scene. I am using HTC vive pro , do you know if there is a plug-in for this in Airsim binaries?

lovettchris, 2021-06-07:
I think you should first look for Unreal support for HTC Vive.  AirSim is just an unreal plugin, and I assume HTC Vive support is also an unreal plugin, so presumably you simply add the 2 plugins together and you get what you want.  So look for some unreal HTC Vive demo environments, then I can show you how to add AirSim to that.

Nilzfar, 2021-06-10:
> I think you should first look for Unreal support for HTC Vive. AirSim is just an unreal plugin, and I assume HTC Vive support is also an unreal plugin, so presumably you simply add the 2 plugins [...]

sorry for my mistake, my headset is HTC Vive Pro. I could use VR mode with the Vive pro in Airsim.NH but I don't know how to get data of eye tracker in binaries.

closed 2021-06-15 by jonyMarino