microsoft/AirSim #3749: Ros noetic TF_REPEATED_DATA

quentingllmt, 2021-05-28:
## Bug report 
- AirSim Version/#commit: latest master
- UE/Unity version: N/A
- autopilot version: N/A
- OS Version: 18.04

### What's the issue you encountered?
When trying to run the airsim ros wrapper in noetic (I need to run it in noetic in order to get to ro1_bridge to works, only on focal) the wrapper starts and works correctly but my logs are full of :
`Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame UrdfBot/odom_local_ned at time 93848.739428 according to authority unknown_publisher`

Related to: https://github.com/ros/geometry2/issues/467

### Settings
N/A

### How can the issue be reproduced?
Run the node in Ros noetic ubuntu 20.04

### Include full error message in text form
ue4-ros       | AirsimROSWrapper Initialized!
ue4-ros       | Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame UrdfBot/odom_local_ned at time 93848.739428 according to authority unknown_publisher
ue4-ros       |          at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
ue4-ros       | Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame UrdfBot/odom_local_ned at time 93848.739428 according to authority unknown_publisher
ue4-ros       |          at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
ue4-ros       | Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame UrdfBot/odom_local_ned at time 93848.739428 according to authority unknown_publisher
ue4-ros       |          at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp

abdo-ameen, 2021-06-06:
I have the same problem.
I use ROS Noetic in order to get Python3 support (Kinetic and Melodic only support Python2 officially) which is needed for PyTorch modules I use in one of the nodes.
Any help from the maintainers would be appreciated.

abdo-ameen, 2021-06-18:
This is getting worse, it's preventing ros services from being executed, even simple services like `airsim_node/drone_1/takeoff` will return `success: False`, and services like `/airsim_node/gps_goal` or `/airsim_node/local_position_goal` will not be executed because their messages are being ignored by `/airsim_node` although the messages have been published to topic `/vel_cmd_world_frame`
[image]
This is the output of `rostopic echo /vel_cmd_world_frame` 
[image]
Moreover, the `/pid_position_node` will loop forever printing out `[PIDPositionController] Moving to goal.` but not moving the drone
[image]

Tao-11-chen, 2022-06-07:
same problem!   is there any solution?