microsoft/AirSim #3762: About the python API control of drone, instructions mismatch

git-liweichao, 2021-06-05:
I have successfully installed UE4 and AirSim, but I have trouble in the control of drone. When I use the python API to move drone, I want to use the API "moveToZAsync(-3, 3)" to move drone to desired position. In fact, it has to achieved position in one second, but it fly about 4 or 5 seconds and then stop. Whether I ignore set something or not? I use the LandscapeMountains environment and UE4.25.4.

clovett, 2021-06-06:
Please include your [settings.json](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md) configuration and your python code.

git-liweichao, 2021-06-07:
hope you can help me. I also find that  when I use the build environment in github of airsim released, I can control rightly according python code.
This is my setting.sjon:
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor"
}

This is my python code:

import airsim
import time

# connect to the AirSim simulator
client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)   # get control
client.armDisarm(True)          # unlock
client.takeoffAsync().join()    # takeoff
print("Takeoff Complete")
# square flight
client.moveToZAsync(-3, 1).join()               
print("Move to (-1, 1) Sucessful")
client.moveToPositionAsync(5, 0, -3, 1).join()  # fly to（5, 0, -3）
print("Move to (5, 0, -1) Sucessful")
client.moveToPositionAsync(5, 5, -3, 1).join()  # fly to（5, 5, -3）
print("Move to (5, 5, -1) Sucessful")
client.moveToPositionAsync(0, 5, -3, 1).join()  # fly to（0, 5, -3）
print("Move to (0, 5, -1) Sucessful")
client.moveToPositionAsync(0, 0, -3, 1).join()  # fly to（0, 0, -3）
print("Move to (0, 0, -1) Sucessful")

client.landAsync().join()       # land
print("Land Sucess!")
client.armDisarm(False)         # lock
client.enableApiControl(False)  # release control

lovettchris, 2021-06-07:
I see, the script works fine, but the drone flys at 1 meter per second, you are asking it to fly 5 meters, so that will take 5 seconds.  To make it fly faster just increase the speed parameter like this:

```
speed = 2 # meters per second
client.moveToZAsync(-3, 1).join()
print("Move to (-1, 1) Successful")
client.moveToPositionAsync(5, 0, -3, speed ).join() # fly to（5, 0, -3）
print("Move to (5, 0, -1) Successful")
client.moveToPositionAsync(5, 5, -3, speed ).join() # fly to（5, 5, -3）
print("Move to (5, 5, -1) Successful")
client.moveToPositionAsync(0, 5, -3, speed ).join() # fly to（0, 5, -3）
print("Move to (0, 5, -1) Successful")
client.moveToPositionAsync(0, 0, -3, speed ).join() # fly to（0, 0, -3）
print("Move to (0, 0, -1) Successful")
```

git-liweichao, 2021-06-09:
Thanks for your answer. But I think it maybe not resolve my question. The fucniton "moveToPositionAsync " , Third parameter is the high I want drone fly. I want to ask it fly 5 meters in 1m/s speed.

lovettchris, 2021-06-09:
I'm talking about the fourth parameter:

```python
def moveToPositionAsync(self, x, y, z, velocity, timeout_sec = 3e+38, drivetrain = DrivetrainType.MaxDegreeOfFreedom, yaw_mode = YawMode(),
        lookahead = -1, adaptive_lookahead = 1, vehicle_name = ''):
```

closed 2021-07-23 by jonyMarino