microsoft/AirSim #3765: getting instance segmentation

meidachen, 2021-06-06:
Hello,

I have been using AirSim for creating synthetic training data for point cloud segmentation. I'm wondering is there a way I can get instance segmentation? To be more specific, the environment I'm creating contains more than 256 objects, I may place the same asset (car model) at different locations in the environment. What I really want to get is each object I'm placing in the environment will have its own label (from 0 to maybe 9999) regardless of its name. I'm also wondering why AirSim only has 256 labels...

I would really appreciate it if anyone knows a workaround for this.

Thank you!

Meida Chen

meidachen, 2021-06-09:
Is there a way to get instance segmentation for more than 256 objects in the environment?

rajat2004, 2021-06-11:
This is a UE4 limitation currently ([Docs](https://docs.unrealengine.com/4.26/en-US/BlueprintAPI/Rendering/SetCustomDepthStencilValue/)), since segmentation uses custom stencil buffer with 1 byte per pixel. See https://github.com/microsoft/AirSim/issues/2252 as well

closed 2021-06-15 by jonyMarino