microsoft/AirSim #3771: RCData had too old timestamp

hjxwhy, 2021-06-08:
## Bug report 
- AirSim Version/#commit: 1.5.0-linux
- UE/Unity version: 4.25
- autopilot version:
- OS Version: ubuntu18.04

### What's the issue you encountered?
I use python to control UAV, I run the demo script of manual_mode_demo.py to send the remote control data "RCData", 
```
client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)

state = client.getMultirotorState()
s = pprint.pformat(state)
print("state: %s" % s)

client.takeoffAsync().join()
client.moveByManualAsync(vx_max = 1E6, vy_max = 1E6, z_min = -1E6, duration = 1E10)
airsim.wait_key('Manual mode is setup. Press any key to send RC data to takeoff')
# timestamp = state.timestamp
client.moveByRC(rcdata = airsim.RCData(pitch = 0.0, throttle = 1.0, is_initialized = True, is_valid = True))

airsim.wait_key('Set Yaw and pitch to 0.5')
# timestamp = client.getMultirotorState().timestamp
client.moveByRC(rcdata = airsim.RCData(roll = 0.5, throttle = 1.0, yaw = 0.5, is_initialized = True, is_valid = True))
```
but the console always has "RCData had too old timestamp", and the UE4 can not stop and have to force close it.

### Settings
just a SimpleFlight

alonfaraj, 2021-06-08:
try to add `timestamp = 0` to your RCData msg

hjxwhy, 2021-06-08:
> try to add `timestamp = 0` to your RCData msg

The default timestamp is 0 and I have tried it, it does not work, I found the program run at `client.moveByManualAsync(vx_max = 1E6, vy_max = 1E6, z_min = -1E6, duration = 1E10)` cause RCData had too old timestamp,  I have annotated the code behind  `client.moveByManualAsync(vx_max = 1E6, vy_max = 1E6, z_min = -1E6, duration = 1E10)`, it appeal RCData had too old timestamp.

alonfaraj, 2021-06-08:
Sorry, you right.

I tried to take a deeper look with debugger.
Although the log says `"RCData had too old timestamp"`, the problem is not the timestamp. 
`RCData` override it and the timestamp is ok.

The problem is `is_valid = false` and then nothing happen in `MultirotorApiBase::moveByManual` which lead to the inaccurate timestamp log msg.
`PawnSimApi::getRCData()` check if your joystick is connected - `joystick_state_.is_valid`, in my case I don't have joystick so it skip the `RCData` msg and I think this is the problem.

If you have joystick you can try verify this theory.

hjxwhy, 2021-06-09:
@alonfaraj Thanks for your reply. You are right, it needs a joystick.

closed 2021-06-09 by hjxwhy