microsoft/AirSim #3776: DepthPerspective is binary, DepthPlanar is grayscale.

h-gokul, 2021-06-09:
## DepthPerspective is binary, DepthPlanar is grayscale.
### Using AirSimNH to collect scene, depth, segmentation in a mono camera setting. 

  

I am sorry if this question is repetitive, I am just few days old in learning AirSim. So can anyone provide the steps clearly on my mistake.

I am using Windows 10, Unreal 4.25.4 

This is my settings.json
```
{
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "SpeedUnitFactor": 1,
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05,
    "Cameras": [
      {
        "CameraName": "0",
        "ImageType": 0,
        "PixelsAsFloat": false,
        "Compress": true
      },
      {
        "CameraName": "0",
        "ImageType": 2,
        "PixelsAsFloat": true,
        "Compress": true
      },
      {
        "CameraName": "0",
        "ImageType": 5,
        "PixelsAsFloat": true,
        "Compress": true
      }
    ]
  },

  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 640,
        "Height": 480,
        "FOV_Degrees": 90,
        "TargetGamma": 2.0
      },
      {
        "ImageType": 2,
        "Width": 640,
        "Height": 480,
        "FOV_Degrees": 90
      },
      {
        "ImageType": 5,
        "Width": 640,
        "Height": 480,
        "FOV_Degrees": 90
      }
    ]
  }
}
```

When I use "DepthPlanar"  by choosing the ImageType: 1, the output looks like grayscale. (Uses blocks environment)

[image]

When I use "DepthPerspective"  by choosing the ImageType: 2, the output looks like DepthVis which results in binary.
[image]

I suspect I am making a mistake in the settings.json file that  is causing this issue, can I get some help for this ?

h-gokul, 2021-06-10:
UPDATE: 
**For the Depth Perspective Image,** after loading pfm files, I got it sorted after **clipping** the image to get proper visualization as suggested in issue #921. Clipping is required since sky is infinitely far away and the floating point values are of order 10^5. 

DepthPerspective Image after clipping 0,100
[image]

DepthPlanner's gray scale issue stays unresolved.

skykim0609, 2021-06-11:
From my experience, I think DepthPerspective image looks binary due to infinite depth along free space and the overall depth image is normalized. Actual depth value is okay.
 Depthplanar topic being grayscale image should be fixed quickly. This bug appears since ver > 1.4.0

closed 2021-06-11 by h-gokul