microsoft/AirSim #3783: Control Speed in Computer Vision mode

h-gokul, 2021-06-10:
## Control Speed in Computer Vision mode
###  Is it possible to control the speed of the player camera in Computer Vision mode ?

### Context:
I am using AirSim using Unreal Engine to collect egomotion data in indoor environments like RealisticRendering.
Indoor environments are small in space and the default set speed of the Computer Vision mode is too high for me to obtain reasonable data.  I am not running any python/C++ scripts (like hello_car.py or cv_mode.py) to collect data but by editing the settings.json file and recording using the GUI in Unreal Editor. Can I get some help me regarding this? 

I have no prior experience with Unreal Engine or AirSim. 

I am using Windows 10 with Unreal 4.25

h-gokul, 2021-06-11:
UPDATE: 

In Plugins/AirSim/Source/ManualController.h the speed_scalar is set  to 1000 by default and reducing it reduces the speed of the pawn. However, the speed of  roll, pitch and yaw motion is still high.

jonyMarino, 2021-06-15:
Hi @h-gokul! Current pre-built environments (v1.5.0) are prior to the addition of that feature (LCtrl to slow down / LShift to speed up.).

h-gokul, 2021-06-15:
Thanks for your response. I was not able to use the pre-built environments for my problem, especially the Building_99 in which "ComputerVision" mode is much more sutiable than the vehicles.

Nevertheless, inability to control the angular speed (roll, pitch and yaw) was a the biggest concern to record data from the indoor environments.

jonyMarino, 2021-06-16:
I can help you to submit a PR with what you need, Gokul!
you can follow this PR: #3221
and extend it for rotation. 
Compare flow for delta_position_ vs. delta_rotation_ in https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/ManualPoseController.cpp

rajrana22, 2021-07-07:
What exactly is the "speed_scaler" value measured in? I know that AirSim uses SI units and Unreal Engine uses cm, but neither of those would make sense as values (too large). Is it meters per second? Is it cm per second? Or is it not a speed but an absolute distance moved with each press of the forward button?

jonyMarino, 2021-07-07:
it is a scaler for incrementing the camera velocity that is in cm/s