microsoft/AirSim #3790: GazeboDrone build error on Mac

nikitabeebe, 2021-06-13:
## Question
### What's your question?
I'm trying to run the GazeboDrone package, following the instructions on the readme. But after I create a new directory build in the edited airsim version, I receive an error with the 'make' command. What am I doing wrong or where is this error coming from? 

### Include context on what you are trying to achieve
Trying to run a gazebo drone simulation with GazeboDrone. 

#### Context details
Most updated version of AirSim, Blocks environment. I installed gazebo 9 version and can open that. Using UE4.26, and running this on a macbook Big Sur 11.2.3. I have been able to get the blocks environment running, as well as gazebo, but I have not yet been successful to run the GazeboDrone in VS due to the make error. 

`nikitabeebe@Nikitas-MBP build % make         
Consolidate compiler generated dependencies of target GazeboDrone
[ 50%] Building CXX object CMakeFiles/GazeboDrone.dir/src/main.cpp.o
In file included from /Users/nikitabeebe/Documents/GitHub/airsim/GazeboDrone/src/main.cpp:21:
In file included from /usr/local/Cellar/gazebo9/9.19.0/include/gazebo-9/gazebo/transport/transport.hh:3:
/usr/local/Cellar/gazebo9/9.19.0/include/gazebo-9/gazebo/transport/Connection.hh:56:65: error: expected class name
    class GZ_TRANSPORT_VISIBLE ConnectionReadTask : public tbb::task
                                                                ^
/usr/local/Cellar/gazebo9/9.19.0/include/gazebo-9/gazebo/transport/Connection.hh:72:20: error: no type named 'task' in namespace 'tbb'
      public: tbb::task *execute()
              ~~~~~^
/usr/local/Cellar/gazebo9/9.19.0/include/gazebo-9/gazebo/transport/Connection.hh:312:61: error: no member named 'allocate_root' in namespace 'tbb::v1::task'
                  ConnectionReadTask *task = new(tbb::task::allocate_root())
                                                 ~~~~~~~~~~~^
/usr/local/Cellar/gazebo9/9.19.0/include/gazebo-9/gazebo/transport/Connection.hh:314:19: error: no type named 'enqueue' in namespace 'tbb::v1::task'; did you mean 'ifqueue'?
                  tbb::task::enqueue(*task);
                  ^~~~~~~~~~~~~~~~~~
                  ifqueue
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/include/net/if_var.h:230:9: note: 'ifqueue' declared here
struct  ifqueue {
        ^
In file included from /Users/nikitabeebe/Documents/GitHub/airsim/GazeboDrone/src/main.cpp:21:
In file included from /usr/local/Cellar/gazebo9/9.19.0/include/gazebo-9/gazebo/transport/transport.hh:6:
/usr/local/Cellar/gazebo9/9.19.0/include/gazebo-9/gazebo/transport/Node.hh:39:58: error: expected class name
    class GZ_TRANSPORT_VISIBLE PublishTask : public tbb::task
                                                         ^
/usr/local/Cellar/gazebo9/9.19.0/include/gazebo-9/gazebo/transport/Node.hh:54:20: error: no type named 'task' in namespace 'tbb'
      public: tbb::task *execute()
              ~~~~~^
/usr/local/Cellar/gazebo9/9.19.0/include/gazebo-9/gazebo/transport/Node.hh:162:52: error: no member named 'allocate_root' in namespace 'tbb::v1::task'
                PublishTask *task = new(tbb::task::allocate_root())
                                        ~~~~~~~~~~~^
/usr/local/Cellar/gazebo9/9.19.0/include/gazebo-9/gazebo/transport/Node.hh:165:17: error: no type named 'enqueue' in namespace 'tbb::v1::task'; did you mean 'ifqueue'?
                tbb::task::enqueue(*task);
                ^~~~~~~~~~~~~~~~~~
                ifqueue
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/include/net/if_var.h:230:9: note: 'ifqueue' declared here
struct  ifqueue {
        ^
8 errors generated.
make[2]: *** [CMakeFiles/GazeboDrone.dir/src/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/GazeboDrone.dir/all] Error 2
make: *** [all] Error 2`

jonyMarino, 2021-06-15:
Hi Nikita. Have you installed `gazebo9` and `libgazebo9`?

nikitabeebe, 2021-06-15:
I've installed gazebo9, but the libgazebo9-dev is the one that gives an error. 
```
Searching for similarly named formulae...
Error: No similarly named formulae found.
Error: No available formula or cask with the name "libgazebo9-dev".
==> Searching for a previously deleted formula (in the last month)...
Error: No previously deleted formula found.
==> Searching taps on GitHub...
Error: No formulae found in taps.
```

jonyMarino, 2021-06-15:
I think I have installed `libgazebo9` not `libgazebo9-dev`.
Can you try installing px4+dependencies with these instructions:
https://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html#gazebo-jmavsim-and-nuttx-pixhawk-targets
?

nikitabeebe, 2021-06-15:
I dont believe that worked. I followed additional instructions for PX4 on mac, but still having the same error for libgazebo9 and libgazebo9-dev. Is this library only functional on Ubuntu?