microsoft/AirSim #3792: Adding Universidad de Zaragoza as collaborator

ppueyor, 2021-06-14:
Hi, as we are using AirSim for our research, we would like to be added to the list to "Who is using AirSim" (https://microsoft.github.io/AirSim/who_is_using.html)

We are a subgroup of ROPERT (Robotics, Perception and Real Time Group) inside of Unizar (Universidad de Zaragoza).
(http://robots.unizar.es)

Our team is called POC (Perception-Oriented Control) team, and our webpage is **https://sites.google.com/unizar.es/poc-team/home**

One work where we used AirSim is for CinemAirSim: https://sites.google.com/unizar.es/poc-team/research/cinematography

Thanks!

jonyMarino, 2021-06-15:
Hi, Pablo. Which link do you want to appear with?

ppueyor, 2021-06-15:
Hi Jony, 

This is the link:
 https://sites.google.com/unizar.es/poc-team/home

closed 2021-06-16 by jonyMarino