microsoft/AirSim #3793: OpenCV C++ and AirSim

pipidx, 2021-06-15:
## Question
### Is it possible to import OpenCV C++ into AirSim project?

jonyMarino, 2021-06-15:
Hi and welcome @pipidx! Yes, you can create an AirSim C++ project that imports/includes the airsim client and OpenCV. 
Please, read the docs:
https://microsoft.github.io/AirSim/apis_cpp.html

pipidx, 2021-06-16:
> 
> 
> Hi and welcome @pipidx! Yes, you can create an AirSim C++ project that imports/includes the airsim client and OpenCV.
> Please, read the docs:
> https://microsoft.github.io/AirSim/apis_cpp.html

pipidx, 2021-06-16:
Thanks for your reply, I have tried to check the official document you suggested, but I was stuck at these below instructions

### Include and Lib Folders

    Include folders: $(ProjectDir)..\AirLib\deps\rpclib\include;include;$(ProjectDir)..\AirLib\deps\eigen3;$(ProjectDir)..\AirLib\include
    => should I include the whole opencv folder into $(ProjectDir)..\AirLib\deps\opencv? 
    
    Dependencies: rpc.lib
    => 'opencv_world411d.lib' is it the file that should be used? 
    
Lib folders: $(ProjectDir)\..\AirLib\deps\MavLinkCom\lib\$(Platform)\$(Configuration);$(ProjectDir)\..\AirLib\deps\rpclib\lib\$(Platform)\$(Configuration);$(ProjectDir)\..\AirLib\lib\$(Platform)\$(Configuration)
    References: Reference AirLib and MavLinkCom to the project references. (Right click your project then go to References, Add reference..., and then select AirLib and MavLinkCom)
    
I tried to follow this structure, but I couldn't understand at all. I tried to put opencv2 folder into $(ProjectDir)..\AirLib\ and it seemed doesn't work. Do we have to include OpenCV file into /deps , /include and /lib separately or in any other way? 

Currently, I'm using Windows 10 and OpenCV 4.1.1 

- I have AirSim folder in :D drive 'D:\AirSim\AirLib\deps'
- And OpenCV in C drive 'C:\OpenCV-4.1.1\opencv'

It would be so pleased if you could briefly show or explain a little bit more about these. Thank you.

pipidx, 2021-06-22:
I have solved already by importing OpenCV C++ library with AirSim. Thank you for your help.

closed 2021-06-22 by jonyMarino