microsoft/AirSim #3799: GPU

EdwinMeriaux, 2021-06-16:
I cannot find any answer to this:
Will there be any issues with AMD GPUs for Airsim?
With the chip shortage and everything it is a bit hard to find good GPUs and if AMD GPUs work that would be great to know.

rajat2004, 2021-06-17:
It would depend on UE4, since AirSim itself doesn't deal with the GPU. UE4 docs or forum threads would be the best place ig.
https://forums.unrealengine.com/t/best-gpu-for-unreal-nvidia-amd/153376 seems to indicate raytracing performance is bad, but benchmarks would be the best way to confirm

EdwinMeriaux, 2021-06-17:
Yes since AMD doesn't do raytracying directly and only recently started something similar.
I won't really mind not have raytracing.
I just was curious if there was any reason why in the Airism Github has recommend specs with Nvidia gpu and I had only seen comments about AMD gpus.

closed 2021-06-18 by EdwinMeriaux