microsoft/AirSim #3800: ERROR: Unable to find plugin 'AirSim', or ERROR: 'common/AirSimSettings.hpp' file not found

xilinzhou, 2021-06-16:
## Question
### What's your question?

I'm trying to run AirSim for the first time (followed instructions [here](https://microsoft.github.io/AirSim/build_windows.html) -- I didn't change anything)

Upon generating the project files and building the Blocks workspace as instructed by the guide, I get `ERROR: Unable to find plugin 'AirSim' (referenced via Blocks.uproject). Install it and try again, or remove it from the required plugin list.`

[image]

I then tried moving `Unreal/Plugins` into the `Blocks` directory, and got the error `'common/AirSimSettings.hpp' file not found` and warnings.

[image]

`/Users/xilinzhou/Documents/GitHub/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirSim.Build.cs: warning: Referenced directory '/Users/xilinzhou/Documents/GitHub/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include' does not exist.
/Users/xilinzhou/Documents/GitHub/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirSim.Build.cs: warning: Referenced directory '/Users/xilinzhou/Documents/GitHub/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/eigen3' does not exist.
/Users/xilinzhou/Documents/GitHub/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirSim.Build.cs: warning: Referenced directory '/Users/xilinzhou/Documents/GitHub/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/MavLinkCom/include' does not exist.
/Users/xilinzhou/Documents/GitHub/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirSim.Build.cs: warning: Referenced directory '/Users/xilinzhou/Documents/GitHub/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include' does not exist.
WARNING: Library '/Users/xilinzhou/Documents/GitHub/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/MavLinkCom/lib/libMavLinkCom.a' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
WARNING: Library '/Users/xilinzhou/Documents/GitHub/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/lib/libAirLib.a' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
WARNING: Library '/Users/xilinzhou/Documents/GitHub/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/librpc.a' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.`

I know the path in the warning doesn't actually exist, but I'm not sure how to get the build to recognize where the plugin and files are.

### Include context on what you are trying to achieve

I'm trying to run AirSim for the first time.

#### Context details

AirSim Version/#commit: 2838 (cloned 06/16/2021)
UE/Unity Version: UE 4.25.4
OS Version: macOS Catalina Version 10.15.2

### Include details of what you already did to find answers
Read https://github.com/Microsoft/AirSim/issues/1235 and https://github.com/microsoft/AirSim/issues/3131

rajat2004, 2021-06-17:
Since you're using macOS, follow the instructions on https://microsoft.github.io/AirSim/build_linux.html

xilinzhou, 2021-06-17:
> Since you're using macOS, follow the instructions on https://microsoft.github.io/AirSim/build_linux.html

Oh no! I'm sorry, I meant to link the Apple/Linux guide, not the Windows one. I've updated the question now.

rajat2004, 2021-06-17:
Okay, thanks for the update. Did the `setup.sh` and `build.sh` from the root AirSim directory complete successfully? Can you run `clean.sh` and then try `build.sh` again and paste the output here? `build.sh` copies the AirLib files to the Unreal/Plugins folder which are showing up as missing in your case

xilinzhou, 2021-06-17:
There's an error showing up in the terminal output near the end of the build (copied below). I've also attached a text file copy of the full build terminal output in case that's helpful ([AirSimBuildOutput.txt](https://github.com/microsoft/AirSim/files/6667291/AirSimBuildOutput.txt)). `setup.sh` seems to be completing successfully.

[ 97%] Linking CXX executable ../output/bin/AirLibUnitTests
Undefined symbols for architecture x86_64:
  "___darwin_check_fd_set_overflow", referenced from:
      mavlinkcom_impl::UdpSocketImpl::pollin(unsigned int) in libMavLinkCom.a(UdpSocketImpl.cpp.o)
      mavlinkcom_impl::UdpSocketImpl::pollout(unsigned int) in libMavLinkCom.a(UdpSocketImpl.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang-8: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [output/bin/AirLibUnitTests] Error 1
make[1]: *** [AirLibUnitTests/CMakeFiles/AirLibUnitTests.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 98%] Linking CXX executable ../output/bin/DroneServer
[ 98%] Built target DroneServer
[100%] Linking CXX executable ../output/bin/DroneShell
[100%] Built target DroneShell
make: *** [all] Error 2

rajat2004, 2021-06-17:
This has been extensively discussed in #2559, so better to continue there itself
Essentially, it's a problem in XCode 11.4, seems to be fixed in 11.4.1 but is present in 11.5 as well, so not completely sure which versions are affected.
You could comment out the lines causing the problem, they're only relevant if using AirSim with ArduPilot. See https://github.com/microsoft/AirSim/issues/2559#issuecomment-615021972 and below for more details

xilinzhou, 2021-06-17:
Worked great, thank you so much!

closed 2021-06-17 by xilinzhou