microsoft/AirSim #3805: How to call ros service

abdo-ameen, 2021-06-20:
## Question
How to call ros services the right way?

### Include context on what you are trying to achieve
I'm just trying to call ros services like `/airsim_node/gps_goal` or `/airsim_node/local_position_goal` but they return errors.

1- First I run my Unreal Blocks environment, press play button
2- launch /airsim_node/ after setting up the terminal
`source devel/setup.bash`
`export DISPLAY=$(cat /etc/resolv.conf | grep nameserver | awk '{print $2}'):0`
`export WSL_HOST_IP=$(cat /etc/resolv.conf | grep nameserver | awk '{print $2}')`
`roslaunch airsim_ros_pkgs airsim_node.launch output:=screen host:=$WSL_HOST_IP`

3- launch `/pid_position_node`
`roslaunch airsim_ros_pkgs position_controller_simple.launch`

4- calling the service with its arguments `rosservice call /airsim_node/gps_goal 5 10 15 1 drone_1` returns `success: False message: ''`

5- calling service `rosservice call /airsim_node/local_position_goal 10 15 20 2 drone_1` returns `ERROR: service [/airsim_node/gps_goal] responded with an error:` and the `/pid_position_node` will loop forever printing out `[PIDPositionController] Moving to goal.` but never moving the drone
[image]

6- calling `/airsim_node/takeoff 1` does move the drone but also return `success: Flase`

I'm I doing something wrong here? I'm I missing something? The documentations provide no more explanations on ros services.

#### Context details
OS: Windows10 with WSL2 running Ubuntu 18.04 and ROS Melodic 
AirSim: 1.5
Python: 3.6
Unreal: 4.26

settings.json:
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ViewMode": "Fpv",
  "ClockSpeed": 1.0,
  "Vehicles": {
    "drone_1": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      },
      "Sensors": {
        "Imu" : {
          "SensorType": 2,
          "Enabled": true
        },
	"Barometer":{
                "SensorType": 1,
                "Enabled": true,
                "PressureFactorSigma": 0.0001825
        },
	"Gps": {
        "SensorType": 3,
        "Enabled" : true,
        "EphTimeConstant": 0.9,
        "EpvTimeConstant": 0.9,
        "EphInitial": 25,
        "EpvInitial": 25,
        "EphFinal": 0.1,
        "EpvFinal": 0.1,
        "EphMin3d": 3,
        "EphMin2d": 4,
        "UpdateLatency": 0.2,
        "UpdateFrequency": 50,
        "StartupDelay": 1
    },
	"Magnetometer": {
        "SensorType": 4,
        "Enabled" : true,
        "NoiseSigma": 0.005,
        "ScaleFactor": 1,
        "NoiseBias": 0,
        "UpdateLatency": 0,
        "UpdateFrequency": 50,
        "StartupDelay": 0
    },
        "LidarCustom": {
          "SensorType": 6,
          "Enabled": true,
          "NumberOfChannels": 16,
          "PointsPerSecond": 10000,
          "X": 0,
          "Y": 0,
          "Z": -1,
          "DrawDebugPoints": true
        }
      },
      "Cameras": {
        "front_center_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 27,
              "DepthOfFieldFstop": 2.8,
              "DepthOfFieldFocalDistance": 200.0, 
              "DepthOfFieldFocalRegion": 200.0,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": 0, "Z": 0.10,
          "Pitch": 0, "Roll": 0, "Yaw": 0
        },
        "front_left_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }, 
            {
              "PublishToRos": 1,
              "ImageType": 1,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": -0.06, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        },
        "front_right_custom": {
          "CaptureSettings": [
            {
              "PublishToRos": 1,
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 0.50, "Y": 0.06, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        }
      },
      "X": 2, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
  },
  "SubWindows": [
    {"WindowID": 0, "ImageType": 0, "CameraName": "front_left_custom", "Visible": false},
    {"WindowID": 1, "ImageType": 0, "CameraName": "front_center_custom", "Visible": false},
    {"WindowID": 2, "ImageType": 0, "CameraName": "front_right_custom", "Visible": false}
  ]
}

### Include details of what you already did to find answers
I tried with different ROS versions, Kinetic, Melodic and Noetic.
I tried with different arguments.

Yousri96, 2021-06-26:
hi, have you found any solutions yet?
i have the same problem

abdo-ameen, 2021-06-27:
Nope, no response from the maintainers yet.

PR #3880 (2021-07-30): fix ROS topic names

Fixes: #3805    
Fixes: #2602
Fixes: #3878 

## About

This PR updates vel_cmd_world_frame and odom_local_ned topic names to include vehicle name so that communication between PIDPositionController and AirsimROSWrapper. The changes for updating the odom_local_ned topic comes from @leoborgnino's work on #2789

## How Has This Been Tested?

ROS node was built with changes applied and the repro steps in #3805 were followed

## Screenshots (if appropriate):
closed 2021-07-30 by jonyMarino