microsoft/AirSim #3806: simGetMeshPositionVertexBuffers() returns meshes with the same vertices

Eastonboy99, 2021-06-21:
Using the simGetMeshPositionVertexBuffers() function in python to get the meshes of objects in the environment results in an object with the same point for its vertices. Getting a cube object (templatecube_48) from the Blocks environment gives me the vertices and indices below. There are also several indices that are repeated. I'm not sure if this is a bug or something I am doing wrong. I am using the example found in meshes.md up until the igl part

```
vertices = [[ 5810. -1650.   148.]
            [ 5810. -1650.   148.]
            [ 5810. -1650.   148.]
            [ 5810. -1650.   148.]
            [ 5810. -1650.   148.]
            [ 5810. -1650.   148.]
            [ 5810. -1650.   148.]
            [ 5810. -1650.   148.]
            [ 5810. -1650.   148.]
            [ 5810. -1650.   148.]]

indices = [[0 0 1]
        [0 2 0]
        [2 0 1]
        [0 3 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]
        [0 0 0]]
```

zimmy87, 2021-07-06:
When I try to run the example script meshes.md to get the MeshPositionVertexBuffersResponse for templatecube_48 in the Blocks environment, I am seeing different output than what you have listed above. Are you trying to run this in the current AirSim master? Also what version of python are you using? I am using Anaconda3 / python 3.8.5

yhabib29, 2022-04-01:
Up !
I have the same issue using latest AirSim v1.7.0 and Unreal Engine 4.25.4.
Anyone managed to retrieve the whole scene mesh ?