microsoft/AirSim #3813: MacOS check failing

jonyMarino, 2021-06-22:
It seems that something changed in GitHub actions and what was only throwing warnings previously now throws errors. 
You can see the error here:
https://github.com/microsoft/AirSim/runs/2876995070

We have tried different solutions in this PR:
https://github.com/microsoft/AirSim/pull/3812
but with no luck. 

Does anyone have an idea on how to fix this?

rajat2004, 2021-06-23:
I've opened #3815 which seems to fix the problem, but additional testing on a macOS system would be for the best. Separate PR since the fix is different from #3812, one can test and compare both separately

jonyMarino, 2021-06-23:
always outstanding Rajat! I guess @saihv can test it?
Thanks!

closed 2021-06-23 by jonyMarino

saihv, 2021-06-24:
Thanks for the fix @rajat2004! Setup and build work fine on my Mac.

PR #3815 (2021-06-23): Remove outdated LLVM tap in OSX

Fixes: #3813    

## About

Removes the `llvm-hs` tap which was causing the failure. Removed since https://formulae.brew.sh/formula/llvm doesn't indicate that the tap is required.

## How Has This Been Tested?

Action CI only. Will be best if someone with macOS machine could test it out as well

## Screenshots (if appropriate):