microsoft/AirSim #3818: How to increase the frequency of C++ API

ClarkZha, 2021-06-23:
## Question
### What's your question?

How to increase the frequency of C++ API, specifically client.simSetVehiclePose().

### Include context on what you are trying to achieve

I am connecting an external drone dynamics simulator to AirSim+Unity. The external dynamics calculates the pose of the vehicle and send it to AirSim via the client.simSetVehiclePose(). Yet significant delay is observed as the client.simSetVehiclePose() is running at 50Hz. 

I cannot bypass the problem via setting the clock-time as the external physics runs with a hardware in the loop. 

Any hint/suggestion is much appreciated!

#### Context details

AirSim Version/#commit: 1.4.0
UE/Unity version: Unity 2019.3.12.f1
OS Version: Ubuntu 18.04

### Include details of what you already did to find answers
Related discussion about using external physics with Unity: #3385, #1828, #3203
Related discussion about API: #3049, #2118

ClarkZha, 2021-06-24:
To add a little bit explanation to the discussion:

I made an experiment that moves the vehicle with `simSetVehiclePose()` and read the pose `simGetVehiclePose()` with at 500Hz. Then I plot out the pose I set and I read as follows:

[image]

I am seeing a 0.02 second delay between when the pose was set and when it is actually updated in the airsim. Moreover, I am seeing the pose being updated at a rate of 100Hz. I am wondering if there is anything I can do to decrease this delay time and increase the pose update frequency.

skykim0609, 2021-06-29:
I am using AirSim for drone simulation, and I speculate that my issue is related to this. I am using airsim ros node and observed significant delay in multirotorrpcclient::getMultirotorState(). Perhaps the reason behind this is that somehow rpc::client::call() function(which also lies behind simSetVehiclePose() function) requires very long time(20ms in my case) to process, which is undesirable behavior.

ClarkZha, 2021-06-29:
> I am using AirSim for drone simulation, and I speculate that my issue is related to this. I am using airsim ros node and observed significant delay in multirotorrpcclient::getMultirotorState(). [...]

Yes I think what you experienced is related to what I observed. Can you please provide more background as this might be helpful with pinning down the problem?  : )

Are you using Unity/Unreal? At what rate are you calling multirotorrpcclient::getMultirotorState() in your ros node and what is the frequency your node is spinning at?

skykim0609, 2021-06-30:
> Yes I think what you experienced is related to what I observed. Can you please provide more background as this might be helpful with pinning down the problem? : )
>
> Are you using Unity/Unreal? At [...]

I am currently using AirSim 1.5.0 with Unreal 4.25 version, and I tried calling getMultirotorState function with 100Hz frequency, because I wanted to get IMU topic in at least 100Hz frequency.(This is done by modifying update_airsim_control_every_n_sec parameter in airsim_node.launch to 0.01) However I observed that the upper limit of topic frequency is about 47Hz, due to time required to process getMultirotorState().

zimmy87, 2021-07-07:
We have checked in the code and the pose is being updated when the API is being called, so the update frequency is the msgpack RPC response frequency. You will need to figure out how to increase that frequency.

closed 2021-07-07 by jonyMarino