microsoft/AirSim #3819: How to change idle thruttle?

EdwinMeriaux, 2021-06-24:
So from this document:
https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/MultiRotorParams.hpp

It says Idle thruttle for simple flight is 50%. Is there any what to change that?

EdwinMeriaux, 2021-06-25:
the drone that I would like to simulate has a particular max thrust and the drone's idle is lower than 50% max thrust. Is there any way to change this?

rajat2004, 2021-07-01:
@ahmed-elsaharti might have a better idea regarding this, referring to his PR https://github.com/microsoft/AirSim/pull/3401

ahmed-elsaharti, 2021-07-01:
Hey guys, 
If I remember correctly, the 50% idle thrust (I'm assuming we're talking about the simpleflight flight controller here) in my case was set by the gamepad/joystick connected to the computer via USB. ie: since i was using a USB gamepad with thumbsticks that default to the center, it pushed out a signal of 50% (ie: center). I remember the idle throttle dropping below 50% once no joysticks were connected to the computer.

Idr exactly where the non-joystick idle throttle was set but I'd assume it would be somewhere in the simpleflight directory since that parameter should be controlled by the autopilot code.

EdwinMeriaux, 2021-07-03:
Hello!

Wait so this idle rate still holds when programming the drones with Python instead of flying with joystick?

EdwinMeriaux, 2021-07-07:
@ahmed-elsaharti

closed 2021-07-07 by jonyMarino