microsoft/AirSim #3825: Problem with AirSim on Ubuntu whith UnrealEngine 4.26

ale99viv, 2021-06-25:
## Question
I, i'm trying to use AirSim(1.5.0) on Ubuntu with Unreal 4.26 but when I try to launch ther project with terminal by the command "./Engine/Binaries/Linux/UE4Editor pathproject/project.uproject" i have this problem. Anyone can halp me? Whith the same method I'm able to use AirSim but wihth Unreal 4.25

### Include details of what you already did to find answers
[image]

ale99viv, 2021-06-28:
what's the name of this .sh? Sorry for my level but on Ubuntu is my first time, before I have used airsim only on Windows.

jonyMarino, 2021-06-28:
It has the name of the environment inside. For Example, for Africa it is Africa_001.sh.
Here you have [the link to the binaries](https://github.com/microsoft/AirSim/releases/tag/v1.5.0-linux)

ale99viv, 2021-06-29:
Thanks for your fast reply @jonyMarino. Yeah I know this but I'm trying to use my own project

jonyMarino, 2021-06-29:
Can you run your project without the AirSim plugin?

wangzhihui22, 2021-07-01:
I also encountered this problem，I cam run my project with the Airsim plugin.

wangzhihui22, 2021-07-01:
> Can you run your project without the AirSim plugin?

I also encountered this problem，I can run my project with the Airsim plugin.

ale99viv, 2021-07-01:
I've solved the problem by using a IDE for building the project. I'm current using Ubntu and so I have do this through QtCreators

wangzhihui22, 2021-07-02:
> Can you run your project without the AirSim plugin?

I also encountered this problem，I can run my project without the Airsim plugin.

ale99viv, 2021-07-02:
@wangzhihui22  What's the error?

jonyMarino, 2021-07-07:
Hi, @wangzhihui22 What's the error you are getting?

closed 2021-07-30 by ale99viv