microsoft/AirSim #3829: add msr::airlib::EarthUtils::GeodeticToNed method

zimmy87, 2021-06-25:
## What feature are you suggesting?
### Overview:

Right now there is a msr::airlib::EarthUtils::GeodeticToNedFast for converting GeoPoints to NED coordinates, but the algorithm implemented is a approximate that can result in errors of up to 10km. There is a corresponding method for converting the opposite way, msr::airlib::EarthUtils::nedToGeodeticFast, that has a non-fast version, but we don't yet have a non-fast version of msr::airlib::EarthUtils::GeodeticToNedFast.

### Smaller Details: 

### Nature of Request:  

- Addition
 

## Why would this feature be useful?
This would remove any possible error in the WorldSimApi::testLineOfSightBetweenPoints API

rajat2004, 2021-06-26:
There's a file [GeodeticConverter.hpp](https://github.com/microsoft/AirSim/blob/master/AirLib/include/common/GeodeticConverter.hpp), it was discussed a bit in https://github.com/microsoft/AirSim/pull/3364#issuecomment-826106220 and later as well. I'll try to bring in the changes I made and see if this turns out to be useful, currently it's unused

zimmy87, 2021-06-29:
Ah I forgot about that file, I've updated calls to msr::airlib::EarthUtils::GeodeticToNedFast with calls to msr::airlib::GeodeticConverter::geodetic2Ned in #3810