microsoft/AirSim #3837: Camera Not Follow Custom Vehicle

rancheng, 2021-06-30:
## Question
### What's your question?
Camera not follow custom vehicle BP. 

### Include context on what you are trying to achieve
I create my own BP with custom mesh and behaviors, I import it with `Pawn_Paths` settings and can successful load it, however, when I hit run, the camera does not follow robot and stays at init position.

#### Context details
Please see the following GIF for detailed information:
[image]

### Include details of what you already did to find answers

zimmy87, 2021-07-07:
Are you able to share your custom BP? We're not able to repro this issue with the `Pawn_Paths` setting locally.

rancheng, 2021-07-29:
Thanks @zimmy87 for your quick reply, its a BP I bought from UE marketplace, a vaccum cleaning machine.

Here is the custom BP:

[Robo.zip](https://github.com/microsoft/AirSim/files/6898091/Robo.zip)
[Robo_part1.zip](https://github.com/microsoft/AirSim/files/6898093/Robo_part1.zip)

Please use the following command to decompress it:
```shell
cat Robo* > Robo_out.zip
unzip Robo_out.zip
```

Simply put the file on your project `content` folder and edit the AirSim settings file like the following:
```json
{
  "SettingsVersion": 1.2, 
  "SimMode": "Car",
  "ViewMode": "SpringArmChase",
  "EnableCollisionPassthrogh": false,
  "EnableCollisions": true,
  "AutoCreate": true,
  "CameraDirector": {
    "FollowDistance": -3,
    "X": 0, "Y": 0, "Z": 0,
    "Pitch": 0, "Roll": 0, "Yaw": 0
  },
  "PawnPaths": 
  { 
    "DefaultCar": {"PawnBP":"Class'/Game/Robo/Blueprints/BP_Robo.BP_Robo_C'"} 
  },
  "Vehicles":{
    "PhysXCar":{
      "VehicleType": "PhysXCar",
      "PawnPath":"DefaultCar",
      "Cameras": { 
        "scene_gt": { 
          "CaptureSettings": [ { "ImageType": 0, "Width": 672, "Height": 376, "FOV_Degrees": 90 } ], 
          "X": 0.00, "Y": 0.00, "Z": 0.00, 
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0 
          } 
        }
    }
  }
}

```

rancheng, 2021-07-29:
After a closer look. I found that the custom BP has updated the pose on the inherited `Body` Mesh component, rather than the base Mesh component compared to the original SUV BP:

This is the custom BP:

[image]

This is the airsim official SUV BP:

[image]

Is there anyway I can change in the setting or code so that I can bind the camera to the `Body` class rather than the Skeleton base mesh?

zimmy87, 2021-09-17:
In your custom BP, have you attached any BP_PIPCamera components to the Body mesh? If that's the mesh that your custom BP is moving, you'll need to attach all BP_PIPCamera components to that one instead of instead of the Mesh root component. I can't currently tell how your custom mesh is configured because I'm getting an error when unzipping it:

```
Archive:  Robo_out.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of Robo_out.zip or
        Robo_out.zip.zip, and cannot find Robo_out.zip.ZIP, period.
```