microsoft/AirSim #3840: Roter  5&6 speed is very low in hexacopter

hhhhhhhhmh, 2021-07-01:
## Question
### What's your question?

Dear all,
I try to change the qualrotor to hexacopter. I  followed https://github.com/microsoft/airsim/wiki/hexacopter
But when I run my project, I found the rotor 5 &6 is very low. it is seems that only 1 rad/s . Please tell my how to fix this.
Thanks
### Include context on what you are trying to achieve

#### Context details

### Include details of what you already did to find answers

rajat2004, 2021-07-01:
Please include details of the platform, AirSim version, exact settings, etc. Also try the settings used in https://github.com/microsoft/AirSim/pull/3402

hhhhhhhhmh, 2021-07-02:
> Please include details of the platform, AirSim version, exact settings, etc. Also try the settings used in #3402

ue4 version : 4.26.2  Airsim version: 1.5.0

jonyMarino, 2021-07-07:
Hi and welcome @hhhhhhhhmh! Can you print the output of getRotorStates() to see if this is a rendering issue? Maybe it is the expected behavior?