microsoft/AirSim #3846: 'Powershell or pwsh not found, please install it' while running 'build.cmd'

AlezzzRd, 2021-07-02:
## Question
### What's your question?
I tried to build the AisSim files with 'build.cmd'. However, it shows 'Powershell or pwsh not found, please install it'. I have installed PowerShell on my computer but the problem still not solved. What should I do?

### Include context on what you are trying to achieve

#### Context details
The version of Visual Studio 2019 Developer Command Prompt is v16.10.3.
The version of AirSim is v1.5.0.

zimmy87, 2021-07-07:
Hi and welcome @AlezzzRd! Can you check which bitness of powershell you have installed and which visual studo 2019 developer command prompt you are running under? You can check the bitness of powershell by [doing this](https://stackoverflow.com/questions/8588960/determine-if-current-powershell-process-is-32-bit-or-64-bit)

ACE199704, 2021-10-10:
I also met that problem...
`**********************************************************************
** Visual Studio 2019 Developer PowerShell v16.11.4
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************
PS C:\Users\RRiverson\source\repos> F:
PS F:\> cd .\AirSim\
PS F:\AirSim> .\build.cmd
Powershell or pwsh not found, please install it.
PS F:\AirSim>
`

What should I do ?

zimmy87, 2022-01-14:
Hi @AlezzzRd and @ACE199704, are either of you still experiencing this issue? If so, can you post the output of running `where powershell` in a Developer Command Prompt?

Fengtao22, 2022-05-15:
Not sure if the issue has already been addressed yet. I had the same problem but solved it by modifying my Visual Studio. Specifically, I only selected "Desktop Development with C++" in the Workloads and chose the newest "Windows 10 SDK" in the Individual Components tab. Besides that, check your environment variable to see if the path to Powershell is there. Hope that helps!

StepLogic, 2022-06-10:
I solved it by add some adding these to my path in  environment variables
[image]

andrz45, 2022-10-07:
> Hi @AlezzzRd and @ACE199704, are either of you still experiencing this issue? If so, can you post the output of running `where powershell` in a Developer Command Prompt?

Hello, I have been trying to solve this problem for two days, and nothing helps, along with colleagues who know a better Windpow itp.. Attempts to assign the exact path in the file "build.cmd" and all sorts of other resolutions. (I do not know exactly what kind, because I was just a follower of friends' instructions) In the end, there was no success. I have the same problem as guys above. To answer your question.  
[image]

cameroncrose72, 2023-03-29:
> I solved it by add some adding these to my path in environment variables [image]

So I found the "System32" powershell path, but not the one in "Program Files" and I'm still getting the error.  Is the one in "Program Files" different?  Or maybe it only needed the system32 one

AhmedSaleh627, 2024-09-01:
I added these to my path in environment variables but still didnt fix it