microsoft/AirSim #3852: Shaky behavior when setting continuous camera orientations

ppueyor, 2021-07-05:
## Bug report 

- AirSim Version/#commit: 1.5
- UE/Unity version: 4.26
- autopilot version:
- OS Version: Ubuntu 20

### What's the issue you encountered?
To avoid aggressive changes in the camera rotations, I implemented a loop where it increments the value of the yaw and pitch softly and calls the function "client.simSetCameraPose()" every 0.05 secs.

The result is that, from time to time, there are glimpses that make the recording very shaky. 

A representative video can be found here: https://drive.google.com/file/d/1TNUsiQthfrNsQh__ObHO0iygBJBGn9y3/view?usp=sharing

### Settings

### How can the issue be reproduced?
Code using C++ API: https://pastebin.com/BbpDcLeW

jonyMarino, 2021-07-07:
Hey Pablo! Can you share a code ready to be compiled? that would make things easier for us. Thanks!

zimmy87, 2021-07-07:
Is it intentional to have the call to `std::this_thread::sleep_for()` outside of your while loop in your code snippet? It looks like there may not be any delay in between your calls to `client.simSetCameraPose()`.

ppueyor, 2021-07-07:
Hi Zimmy. It is intentional. It is inside a thread that moves the gimbal of the camera softly to a determinate value of pitch, yaw. I want to have a soft camera rotation that increments slowly every X milliseconds. In this way, the footage will look continuous.

Jony, I created a demo code where it is easy to reproduce this. The gimbal stabilization is set to true in the settings, maybe it is related to this problem.

Code: https://pastebin.com/SLHDAkQi
Video (more noticeable from sec 13): https://drive.google.com/file/d/1EEit30PeFTEckco6hxEdxgvONpx5Bs6C/view?usp=sharing

Thanks for answering!

ppueyor, 2021-08-29:
Hey guys!
Any insights about this bug?

ppueyor, 2021-08-30:
Hi again guys, 

I found the bug that was causing that problem. It was an if that was just set for the gimbal stabilization mode in PIPCamera.cpp.

To fix it, it is as simple as to add the closing brackets to that line (281-286):

> if (gimbal_stabilization_ > 0) {
        gimbald_rotator_.Pitch = rotator.Pitch;
        gimbald_rotator_.Roll = rotator.Roll;
        gimbald_rotator_.Yaw = rotator.Yaw;
    }
    this->SetActorRelativeRotation(rotator);

to 

> if (gimbal_stabilization_ > 0) {
        gimbald_rotator_.Pitch = rotator.Pitch;
        gimbald_rotator_.Roll = rotator.Roll;
        gimbald_rotator_.Yaw = rotator.Yaw;
    }
    else{
        this->SetActorRelativeRotation(rotator);
    }   

If you want, I can create a PR with this fix.

jonyMarino, 2021-08-30:
Hi Pablo! yes, sure, a PR is welcome!

ppueyor, 2021-09-23:
Hi Jony, 

I created a new PR with this minor fix. PR #4037 .

closed 2021-09-23 by ppueyor