microsoft/AirSim #3862: Cannot connect to standalone custum environment

mmuetert, 2021-07-07:
## Question
### What's your question?

I run the shell script to start my custom environment and it seems to start properly. Then I start my c++ code to control the quad copter, and the code says it's connected, but nothing happens. The drone doesn't move, the props don't spin. When I use the UE editor to run the environment, everything seems to work fine, but the playback is stuttery. I can interface with the standalone version of the blocks environment just fine.

### Include context on what you are trying to achieve

#### Context details

OS: Ubuntu 18.04
Airsim: newest
UE 4.24

### Include details of what you already did to find answers
I tried repackaging the environment and the same thing happens.

rajat2004, 2021-07-10:
Could you post the script you're using the control the vehicle? Also, check if Car and Computervision vehicles are working as well

jonyMarino, 2021-07-16:
Hi @mmuetert! Also share your settings.json, please