microsoft/AirSim #3866: AirSim with ROS WSL2 - AirSim node process has died just after being connected.

imanf94, 2021-07-08:
## Question
I have a similar problem to issue #3667. I am working on connecting ROS (in WSL2) to AirSim on windows 10. My AirSim environment is the City latest release (v1.5). The ROS on WSL2/Ubuntu 18 is working properly. When I try to run this code `roslaunch airsim_ros_pkgs airsim_node.launch output:=screen host:=$WSL_HOST_IP` the connection established successfully but I get this error which led the process to die. What is the problem?

## Error
oslaunch airsim_ros_pkgs airsim_node.launch output:=screen host:=$WSL_HOST_IP
... logging to /home/iman/.ros/log/874c932c-dfb8-11eb-8f90-00155d81e8b6/roslaunch-DESKTOP-4BJ7Q09-377.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://DESKTOP-4BJ7Q09:40949/

SUMMARY
PARAMETERS

/airsim_node/host_ip: 172.30.64.1
/airsim_node/is_vulkan: False
/airsim_node/publish_clock: False
/airsim_node/update_airsim_control_every_n_sec: 0.01
/airsim_node/update_airsim_img_response_every_n_sec: 0.05
/airsim_node/update_lidar_every_n_sec: 0.01
/rosdistro: melodic
/rosversion: 1.14.11
NODES
/
airsim_node (airsim_ros_pkgs/airsim_node)
ned_to_enu_pub (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

process[airsim_node-1]: started with pid [392]
process[ned_to_enu_pub-2]: started with pid [393]
Waiting for connection - X
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
terminate called after throwing an instance of 'rpc::rpc_error'
what(): rpc::rpc_error during call
[airsim_node-1] process has died [pid 392, exit code -6, cmd /home/iman/AirSim/ros/devel/lib/airsim_ros_pkgs/airsim_node __name:=airsim_node __log:=/home/iman/.ros/log/874c932c-dfb8-11eb-8f90-00155d81e8b6/airsim_node-1.log].
log file: /home/iman/.ros/log/874c932c-dfb8-11eb-8f90-00155d81e8b6/airsim_node-1*.log

## Settings.json
And here is my settings.json files which are located in Documents/AirSim folder of both Windows 10 and WSL/Ubuntu. I should mention that my WSL2/Ubuntu didn't have the settings.json and even the folder. So I added the file manually:

```

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "ViewMode": "SpringArmChase",
  "PhysicsEngineName": "FastPhysicsEngine",
  "ClockType": "SteppableClock",
  "ClockSpeed": 0.34,
  "Vehicles": {
    "drone_1": {
      "VehicleType": "PhysXCar",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": true,
      "EnableCollisions": false,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      },
      "Sensors": {
        "LidarSensor": {
          "SensorType": 6,
          "Enabled": true,
          "NumberOfChannels": 16,
          "Range": 30,
          "RotationsPerSecond": 10,
          "PointsPerSecond": 10000,
          "X": 0,
          "Y": 0,
          "Z": -1,
          "Roll": 0,
          "Pitch": 0,
          "Yaw": 0,
          "VerticalFOVUpper": 5,
          "VerticalFOVLower": -5,
          "HorizontalFOVStart": 180,
          "HorizontalFOVEnd": -180,
          "DrawDebugPoints": false,
          "DataFrame": "SensorLocalFrame"
        },
	"Imu" : {
          "SensorType": 2,
          "Enabled": true
        }
      },
     "Cameras": {
        "leftcamera_1": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 90,
              "TargetGamma": 1.5
            }
          ],
          "X": 0, "Y": -0.06, "Z": -2.4,
          "Pitch": 0, "Roll": 0, "Yaw": 0
        },
        "rightcamera_1": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 640,
              "Height": 480,
              "FOV_Degrees": 90,
              "TargetGamma":  @1.5
            }
          ],
          "X": 0, "Y": 0.06, "Z": -2.4,
          "Pitch": 0, "Roll": 0, "Yaw": 0
        }
      },
      "X": 0, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
  },
  "SubWindows": [
    {"WindowID": 1, "ImageType": 0, "CameraName": "leftcamera_1", "Visible": false},
    {"WindowID": 2, "ImageType": 0, "CameraName": "rightcamera_1", "Visible": false}
  ]
}

```

jonyMarino, 2021-07-16:
Hi and welcome @imanf94! Please try disabling your firewall and see if that fixes the issue.

imanf94, 2021-11-03:
Hi @jonyMarino. After struggling a while I solved the issue. Disabling the firewall couldn't help. However instead of using standalone City environment release for Windows I used the Blocks environment in Unity. I think this was a bug in CityEnviron v1.4.0.

closed 2021-11-05 by imanf94