microsoft/AirSim #3868: Too many undeclared identifier errors when building Airsim on mac

sritank, 2021-07-10:
## Question
### What's your question?

I'm trying to build the Airsim repo on my Macbook Pro macOS Big Sur 11.4 with the instructions [here ](https://microsoft.github.io/AirSim/build_linux#build-airsim). The setup.sh script completes successfully (log attached) and identifies that I have all the necessary packages. However, when I try to build the repo using the build.sh script I get too many errors stating that I have undeclared identifiers (error log attached). I have clang 8 and llvm version 8 installed. 

My Xcode version is 12.5.1. 

Unreal Engine version 4.25.4

I've tried searching for the specific error messages like 
_"error: unknown type name 'uint32_t'
                uint32_t param_uint32;"_
They weren't helpful in fixing the issue as I can't make any code changes and I don't know if it has anything to do with gcc/g++
[build_out.txt](https://github.com/microsoft/AirSim/files/6794748/build_out.txt)
[setup_out.txt](https://github.com/microsoft/AirSim/files/6794749/setup_out.txt)
. Can someone please help me with what I am missing here?
### Include context on what you are trying to achieve

#### Context details

### Include details of what you already did to find answers

rajat2004, 2021-07-10:
The problem looks to be with LLVM itself and not AirSim. I guess it's happening due to older clang not supported on OSX 11. Could you try a newer clang such as 11 or 12? https://github.com/microsoft/AirSim/issues/3691 might be helpful. In case it works, the documentation and steps should be updated, but it'll be a bit difficult to test

sritank, 2021-07-10:
I tried the steps in #3691 by installing latest llvm but for some reason it still looks for llvm@8. My clang version is 12.0.1
Homebrew clang version 12.0.1
Target: x86_64-apple-darwin20.5.0
Thread model: posix
InstalledDir: /usr/local/opt/llvm/bin

I changed the directory in line 54 as specified. My build.sh file is attached below along with the log when I run ./build.sh

Why does it still look for llvm@8 when I have clearly uninstalled it? When ./build.sh prints out the clang it is using there is no llvm@8 there
[build_sh_output.txt](https://github.com/microsoft/AirSim/files/6794808/build_sh_output.txt)

[build_sh.txt](https://github.com/microsoft/AirSim/files/6794803/build_sh.txt)

rajat2004, 2021-07-10:
That's strange, try running `clean.sh` once before `build.sh`, it might be using the earlier CMake cache

sritank, 2021-07-10:
That was the issue. Now I'm at the same point as #3691 where the Xcode build fails with exit code 6 (multi touch). Since the main issue here is resolved, I'll close this.

Thanks a lot @rajat2004 !

closed 2021-07-10 by sritank