microsoft/AirSim #3869: Lidar flickering

StevenCoral, 2021-07-10:
## Question
### What's your question?

When adding lidar to my (car) vehicle, i get these flickerings. It is as if an "empty" part of the radar is scanning around, repeating in unknown intervals.  
I figure it has something to do with the way the points are calculated, but it doesn't matter what combination I try, the flicker still occurs but at different widths / intervals.  
As someone who works with lidars a lot, I know this should not be happening. There must be some combination of angles, points per seconds, channels etc. which mitigates this.  

### Include context on what you are trying to achieve

#### Context details

I'm using Windows 10 with latest Airsim version and python 3.8.4.
You can see the screenshots and could not find how to upload the json file, but know that it happens with any combination of lidar properties that I have tried.
I am trying to achieve a solid stream of constant points.  

[image]

### Include details of what you already did to find answers
Could not find a similar question yet.

rajat2004, 2021-07-11:
Pretty strange, my first guess was that maybe some part of the vehicle itself was present in front of the lidar, and hence causing the missing section. But that area is different in the third image, so probably not the case. Have you tried moving the Lidar to different positions, let's say 1m above the vehicle, or a bit to the front, etc? 
To share the settings, just copy-paste the settings.json as a code block here, or could upload it to GDrive or something else and share the link here as well. If you're interested, the Lidar part of the code is in [here](https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/UnrealSensors/UnrealLidarSensor.cpp), it could probably be improved though

StevenCoral, 2021-07-11:
> Pretty strange, my first guess was that maybe some part of the vehicle itself was present in front of the lidar, and hence causing the missing section. But that area is different in the third image, [...]

Actually, I did put it in different locations, and also with different angular ranges. Imagine that this "missing slice" is rotating around the vehicle. So if I define a full 360 circle, I have a spinning missing slice. If I define only a section, then this weird thing appears every now and then. The behavior changes a bit with different numbers, but still exists.
The funny thing is that a similar thing happens with a real Velodyne and ROS driver if you are not using it correctly (and the workaround cannot be utilized here because the mechanism is different).
I'll take a look at the lidar code though I doubt I could muster up anything unless people wouldn't mind that the scanning pattern will be overriden (which is what happens with the ROS drivers for lidars anyway, the whole pointcloud is published every 1/FPS rather than "the points that were scanned so far").

My settings.json is on a different computer, maybe i'll copy it later today.

Ryton, 2021-10-01:
Might it have to do with the alg capping the number points to a maximum of 1E5 points, due to performance?

With your settings
                    "NumberOfChannels": 32,
                    "RotationsPerSecond": 10,
                    "PointsPerSecond": 200000,
With a divison by 360 (degrees), and with your rotations per second and by number of line that results in it a target of 115200 so about 15% excess is cut off.
You could try to either reduce the rotations per second to below 8, reduce the number of lines, or set a value, or increase the hard limit (1E5) in the airsim code. (i forgot where it was, will look it up)

StevenCoral, 2021-10-04:
> Might it have to do with the alg capping the number points to a maximum of 1E5 points, due to performance? IN UnrealLidarSensor.cpp, line 59 and beyond `constexpr float **MAX_POINTS_IN_SCAN = 1e+5f**; ... uint32 total_points_to_scan = FMath::RoundHalfFromZero(params.points_per_second * delta_time); if (total_points_to_scan > MAX_POINTS_IN_SCAN) { total_points_to_scan = MAX_POINTS_IN_SCAN; UAirBlueprintLib::LogMessageString("Lidar: ", "Capping number of points to scan", LogDebugLevel::Failure); } `
> 
> With your settings "NumberOfChannels": 32, "RotationsPerSecond": 10, "PointsPerSecond": 200000, With a divison by 360 (degrees), and with your rotations per second and by number of line that results in it a target of 115200 so about 15% excess is cut off. Normally that message should be present in the log.
> 
> => Solution: You could either reduce the rotations per second to below 8, reduce the number of lines, or set a value, or increase the hard limit (1E5) in the airsim code. (i forgot where it was, will look it up)

Thank you, I already worked around it. it also happens with less than 1e5 points, and depending on the setting, it might also be a "double-scanned" slice instead of a missing one. I have gone through the code, and I hate to say it but the mechanism is flawed. I wonder how nobody noticed it before. Don't get me wrong, I do appreciate the hard work being put into the plugin, but unfortunately I do not have the time to fix the sensor due to consistent overload. 
Regardless, people should know that lidars (or at least velodyne) are not defined by the same parameters as in AirSim, nor do their drivers behave in that manner.

zhangmanCN, 2021-12-30:
Hi,I also meet this trouble.If anyone have some way to solve it ,please discuss it

zhangmanCN, 2022-01-04:
Hello, although I haven't understood the mechanism, I think I have found a solution. The lack of radar data seems to be related to the UE4 simulation frame rate. When the simulation frame rate is greater than the radar scanning frequency, the radar data will be lost. It may be that the refresh of the simulation frame will cut off the radar data. An effective solution is to set the simulation frame rate to half of the radar frequency (if the radar is 10Hz, set the UE4 simulation frame rate to 5Hz), so as to ensure that the radar can obtain complete data. The setting method is to press the "~" keyboard on the simulation page, and then enter "t.maxFPS 5"