microsoft/AirSim #3870: Cannot compile Blocks environment on Ubuntu 18.04

nikitabeebe, 2021-07-10:
## Question
### What's your question?
I am trying to run the basic blocks environment on UE 4.25.4 with my Ubuntu 18.04 and latest release of Airsim. I can successfully compile Airsim, and can open up UE4 through the UnrealEngine binary folder within Arisim. However when I go to browse for the blocks environments, I receive a warning either that this has been built with a different engine and asking if I want to make a copy, or other times I receive an error 'Missing Blocks Module' that the following modules are built with a different engine (Blocks, Airsim) and asking if I want to rebuild. Neither of these options work, and Unreal either briefly loads Blocks then my entire computer freezes screen, I receive the warning it cannot compile and to open in Null source code, or that Engine modules are out of date and cannot be compiled, try building with IDE. 

I have tried compiling with QTcreator, but am now receiving an error that 
g++: error trying to exec '/usr/lib/gcc/x86_64-linux-gnu/7/cc1plus': execv: Argument list too long
g++: error trying to exec 'as': execvp: Argument list too long
make: *** [AITestSuite.o] Error 1

### Include context on what you are trying to achieve
N/A

#### Context details
Ubuntu 18.04, UE 4.25.4
All default settings

### Include details of what you already did to find answers
I have recloned Airsim, tried running through command line, installed Vulkan as I initially I had an issue and was forced to use -opengl4 (though that no longer works).

Any thoughts on where I might have gone wrong? Everything compiles with .setup, .build, .GenerateProjectFiles without issue up until trying to run Blocks.

SamPusegaonkar, 2021-07-13:
Are you sure you are cloning and building the right version of Unreal? I had the same issue while building as I installed unreal from their repo and not the exact specific version.

Try building again using this: 
git clone -b 4.25 https://github.com/EpicGames/UnrealEngine.git
cd UnrealEngine
./Setup.sh
./GenerateProjectFiles.sh
make

(Source: https://microsoft.github.io/AirSim/build_linux)

nikitabeebe, 2021-07-14:
Thanks, I'll give that a try. I thought I downloaded the right version, but I'll re-download from the instructions on AirSim. Fingers crossed that was the only issue!

Astaxuqichao, 2021-09-10:
i use Unreal 4.27, and encounter the same situation.

bozcani, 2021-12-16:
I was getting similar message on Ubuntu 20.04 with Unreal 4.27.1. When I opened "Blocks.uproject" in Editor, I got the following message: 
`Engine modules are out of date and cannot be compiled, try building with IDE.`

It is solved after building the environment from the terminal:

`$ <path to Unreal Engine>/Engine/Build/BatchFiles/Linux/Build.sh Linux Development -Project=<path to Blocks project>/Blocks.uproject -TargetType=Editor`

talha-opteran, 2021-12-16:
Thanks @bozcani. For anyone using VSCode, I also managed to get it run by opening the VSCode workspace file generated when trying to open the `Blocks.uproject` file with the UE editor. In VSCode, press `Ctrl + Shift + B` to open menu for building tasks and run the task for `BlocksEditor Linux Development Build`. This essentially runs the same command as above.

zimmy87, 2022-01-14:
Hey @nikitabeebe, are you still experiencing the issue described in your original post?

zimmy87, 2022-02-02:
Closing this as it looks like this has been resolved.