microsoft/AirSim #3874: Gazebo Drone 'make' command fails

nikitabeebe, 2021-07-12:
## Question
### What's your question?
When following the readme for running the GazeboDrone project, I receive an error following the 'make' command that 'recipe for target all' fails. Am I missing a dependency or path with the cmake?

### Include context on what you are trying to achieve
I am trying to run gazeboDrone after having installed Gazebo and AirSim Blocks binary. 

#### Context details
Ubuntu 18.04, newest version of Airsim, Blocks Binary, UE4.25.4
I have modified the setting.json file to include External Physics Engine, and I have changed the .build.sh file to be gcc and g++, then compiled that successfully. 

My full error message is quite long, but this is the beginning and end:
summer-dev@summerdev-desktop:~/repos/airsim/GazeboDrone/build$ make 
```
[ 50%] Building CXX object CMakeFiles/GazeboDrone.dir/src/main.cpp.o
[100%] Linking CXX executable GazeboDrone
../../AirLib/lib/x64/librpc.a(client.cc.o): In function `client':
/home/summer-dev/repos/airsim/external/rpclib/rpclib-2.3.0/lib/rpc/client.cc:174: undefined reference to `std::__1::to_string(int)'
../../AirLib/lib/x64/librpc.a(client.cc.o): In function `clmdep_asio::ip::basic_resolver_query<clmdep_asio::ip::tcp>::basic_resolver_query(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, clmdep_asio::ip::resolver_query_base::flags)':
/home/summer-dev/repos/airsim/external/rpclib/rpclib-2.3.0/dependencies/include/asio/ip/basic_resolver_query.hpp:150: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)'
```

/home/summer-dev/repos/airsim/external/rpclib/rpclib-2.3.0/dependencies/src/format.cc:1330: undefined reference to `std::__1::basic_ostream<char, std::__1::char_traits<char> >::write(char const*, long)'
collect2: error: ld returned 1 exit status
CMakeFiles/GazeboDrone.dir/build.make:214: recipe for target 'GazeboDrone' failed
make[2]: *** [GazeboDrone] Error 1
CMakeFiles/Makefile2:82: recipe for target 'CMakeFiles/GazeboDrone.dir/all' failed
make[1]: *** [CMakeFiles/GazeboDrone.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2
```

```

``

### Include details of what you already did to find answers

lovettchris, 2021-07-15:
Sorry I’m not an expert with the GazeboDrone setup.  But I did just follow the GazeboDrone readme myself on Ubuntu, building AirSim with gcc and g++ allows the GazeboDrone build to succeed:
```
Scanning dependencies of target GazeboDrone
[ 50%] Building CXX object CMakeFiles/GazeboDrone.dir/src/main.cpp.o
[100%] Linking CXX executable GazeboDrone
[100%] Built target GazeboDrone
```
and this works:
```
./GazeboDrone
Waiting for connection - XXXX...
```

Usually an undefined reference to something like this `std::__1::basic_ostream<char, std::__1::char_traits >::write(char const*, long)` means there is a C++ compiler mismatch in your build somewhere.  Like something still got built with clang and something got built with gcc and their std c++ libraries don’t match.

So when you build AirSim make sure the build.sh outputs something like this indicating cmake is really choosing to use gcc and g++.

```
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/g++
-- Check for working CXX compiler: /usr/bin/g++ -- works
```

You may need to completely remove `build_release` to force Cmake to notice the change you made to `build.sh`.  You could also try doing a “git clean -dfx” to restore your repo to clean state, then try your gcc build again and see if it works better…

nikitabeebe, 2021-07-16:
Removing the build_release folder resolved the issue. Thank you!

closed 2021-07-16 by nikitabeebe

HarrySoteriou, 2022-03-04:
Specs:
Windows 10
Unreal Engine 4.27.2
WSL2 and running VcXsrv (in case this problem has to do with graphics)

Context:
I can run AirSim on Windows and I can also run the ros wrapper and PX4 throught wsl2 (Meaning wsl2 doesnt seem to be the problem). When it comes to building gazebo as my Physics Engine I can't get it to run.

Problem:
I have build AirSim 2 times on wsl2: once using clang compiler and once using  gcc 8.4
cmake -DCMAKE_C_COMPILER=gcc-8 -DCMAKE_CXX_COMPILER=g++-8 ..
but when I try to execute "make" I get the following error:

Error log:
make[2]: *** [CMakeFiles/GazeboDrone.dir/build.make:149: GazeboDrone] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/GazeboDrone.dir/all] Error 2
make: *** [Makefile:91: all] Error 2