microsoft/AirSim #3877: Geographic area can not be selected in docker setup ubuntu

omiderfanmanesh, 2021-07-13:
I am trying to run AirSlim on docker and when I run this command 
```python

python3 build_airsim_image.py --base_image=nvidia/cudagl:11.4.0-devel-ubuntu20.04 --target_image=airsim_binary:11.4.0-devel-ubuntu20.04

```
the log is : 
```
root@omid:/home/omid/OMID/projects/python/AirSim-master/docker# python3 build_airsim_image.py --base_image=nvidia/cudagl:11.4.0-devel-ubuntu20.04 --target_image=airsim_binary:11.4.0-devel-ubuntu20.04
docker build --network=host -t airsim_binary:11.4.0-devel-ubuntu20.04 -f Dockerfile_binary --build-arg BASE_IMAGE=nvidia/cudagl:11.4.0-devel-ubuntu20.04 .
Sending build context to Docker daemon   12.8kB
Step 1/14 : ARG BASE_IMAGE=nvidia/cudagl:10.0-devel-ubuntu18.04
Step 2/14 : FROM $BASE_IMAGE
 ---> ab2322da44da
Step 3/14 : RUN apt-get update
 ---> Using cache
 ---> 2ad09ba09655
Step 4/14 : RUN apt-get install 	python3 	python3-pip 	sudo 	libglu1-mesa-dev 	xdg-user-dirs 	pulseaudio sudo 	x11-xserver-utils 	-y --no-install-recommends
 ---> Running in a4c1379adb3a
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  dbus dbus-user-session dconf-gsettings-backend dconf-service glib-networking
  glib-networking-common glib-networking-services gsettings-desktop-schemas
  iso-codes libapparmor1 libargon2-1 libasound2 libasound2-data
  libasound2-plugins libasyncns0 libbrotli1 libcap2 libcap2-bin
  libcryptsetup12 libdbus-1-3 libdconf1 libdevmapper1.02.1 libflac8
  libglu1-mesa libgssapi-krb5-2 libgstreamer-plugins-base1.0-0
  libgstreamer1.0-0 libice6 libicu66 libip4tc2 libjack-jackd2-0 libjson-c4
  libjson-glib-1.0-0 libjson-glib-1.0-common libk5crypto3 libkeyutils1
  libkmod2 libkrb5-3 libkrb5support0 libltdl7 libmpdec2 libogg0 liborc-0.4-0
  libpam-systemd libproxy1v5 libpsl5 libpulse0 libpulsedsp libpython3-stdlib
  libpython3.8-minimal libpython3.8-stdlib libsamplerate0 libsm6
  libsnapd-glib1 libsndfile1 libsoup2.4-1 libsoxr0 libspeexdsp1 libsystemd0
  libtdb1 libvorbis0a libvorbisenc2 libwebrtc-audio-processing1 libwrap0
  libxaw7 libxcursor1 libxi6 libxml2 libxmu6 libxmuu1 libxpm4 libxrandr2
  libxrender1 libxt6 libxtst6 mime-support pulseaudio-utils python-pip-whl
  python3-distutils python3-lib2to3 python3-minimal python3-pkg-resources
  python3-setuptools python3-wheel python3.8 python3.8-minimal systemd
  systemd-sysv systemd-timesyncd tzdata x11-common
Suggested packages:
  isoquery alsa-utils krb5-doc krb5-user libvisual-0.4-plugins
  gstreamer1.0-tools jackd2 snapd udev pavumeter pavucontrol paman paprefs
  ubuntu-sounds avahi-daemon python3-doc python3-tk python3-venv
  python-setuptools-doc python3.8-venv python3.8-doc binfmt-support
  systemd-container policykit-1 nickle cairo-5c xorg-docs-core
Recommended packages:
  alsa-ucm-conf alsa-topology-conf libpam-cap dmsetup
  gstreamer1.0-plugins-base krb5-locales publicsuffix file rtkit python3-dev
  networkd-dispatcher libnss-systemd
The following NEW packages will be installed:
  dbus dbus-user-session dconf-gsettings-backend dconf-service glib-networking
  glib-networking-common glib-networking-services gsettings-desktop-schemas
  iso-codes libapparmor1 libargon2-1 libasound2 libasound2-data
  libasound2-plugins libasyncns0 libbrotli1 libcap2 libcap2-bin
  libcryptsetup12 libdbus-1-3 libdconf1 libdevmapper1.02.1 libflac8
  libglu1-mesa libglu1-mesa-dev libgssapi-krb5-2
  libgstreamer-plugins-base1.0-0 libgstreamer1.0-0 libice6 libicu66 libip4tc2
  libjack-jackd2-0 libjson-c4 libjson-glib-1.0-0 libjson-glib-1.0-common
  libk5crypto3 libkeyutils1 libkmod2 libkrb5-3 libkrb5support0 libltdl7
  libmpdec2 libogg0 liborc-0.4-0 libpam-systemd libproxy1v5 libpsl5 libpulse0
  libpulsedsp libpython3-stdlib libpython3.8-minimal libpython3.8-stdlib
  libsamplerate0 libsm6 libsnapd-glib1 libsndfile1 libsoup2.4-1 libsoxr0
  libspeexdsp1 libtdb1 libvorbis0a libvorbisenc2 libwebrtc-audio-processing1
  libwrap0 libxaw7 libxcursor1 libxi6 libxml2 libxmu6 libxmuu1 libxpm4
  libxrandr2 libxrender1 libxt6 libxtst6 mime-support pulseaudio
  pulseaudio-utils python-pip-whl python3 python3-distutils python3-lib2to3
  python3-minimal python3-pip python3-pkg-resources python3-setuptools
  python3-wheel python3.8 python3.8-minimal sudo systemd systemd-sysv
[... 448 lines cut ...]
invoke-rc.d: could not determine current runlevel
invoke-rc.d: policy-rc.d denied execution of start.
Setting up libwebrtc-audio-processing1:amd64 (0.3.1-0ubuntu3) ...
Setting up libflac8:amd64 (1.3.3-1build1) ...
Setting up libkrb5support0:amd64 (1.17-6ubuntu4.1) ...
Setting up tzdata (2021a-0ubuntu0.20.04) ...
debconf: unable to initialize frontend: Dialog
debconf: (TERM is not set, so the dialog frontend is not usable.)
debconf: falling back to frontend: Readline
Configuring tzdata
------------------

Please select the geographic area in which you live. Subsequent configuration
questions will narrow this down by presenting a list of cities, representing
the time zones in which they are located.

  1. Africa      4. Australia  7. Atlantic  10. Pacific  13. Etc
  2. America     5. Arctic     8. Europe    11. SystemV
  3. Antarctica  6. Asia       9. Indian    12. US
Geographic area: 8
```

The problem is that when I want to write 8 in front of ```Geographic area``` and then press enter, nothing will be executed and it is just going to a new line in the console. 

My Ubuntu version is 20.10.

jonyMarino, 2021-07-20:
Hi. Can You try  with an Ubuntu 18.04 image?

ale99viv, 2021-07-22:
Did use Ubuntu 18.04 resolve this problem?

omiderfanmanesh, 2021-07-22:
no this is work for me 
```
python build_airsim_image.py \
   --base_image=nvidia/cudagl:10.0-devel-ubuntu18.04 \
   --target_image=airsim_binary:10.0-devel-ubuntu18.04
```

jonyMarino, 2021-07-27:
Can we close this, or are you interested in using an ubuntu 20.04 image?

ale99viv, 2021-07-27:
I have had the same problem and I solved by follow this(https://github.com/microsoft/AirSim/issues/3703). Follow this and you will able to use 20.04 image and vulkan.

Alfr3do, 2021-10-20:
It would be great to use 20.04, @ale99viv link looks like something different.

tommy-neeld-rft, 2022-11-17:
Had this issue, got around it by creating a variable to hold the timezone and passing it into an `/etc/timezone` file (which I think gets picked up by python)
```
FROM ubuntu:20.04

# export timezone - for python3.9-dev install
ENV TZ=Europe/London

# place timezone data /etc/timezone
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone

# install
RUN apt-get update && apt-get -y install --no-install-recommends python3.9-dev

```

dvogureckiy99, 2023-01-24:
> Had this issue, got around it by creating a variable to hold the timezone and passing it into an `/etc/timezone` file (which I think gets picked up by python)
>
> ```
> FROM ubuntu:20.04
>
> # export [...]
I don't understand and have got this 
```
docker/build_docker.sh: line 3: FROM: command not found
docker/build_docker.sh: line 5: ENV: command not found
docker/build_docker.sh: line 6: RUN: command not found
docker/build_docker.sh: line 8: RUN: command not found
docker/build_docker.sh: line 9: RUN: command not found

```

dangost, 2023-10-13:
> I don't understand and have got this
>
> ```
> docker/build_docker.sh: line 3: FROM: command not found
> docker/build_docker.sh: line 5: ENV: command not found
> docker/build_docker.sh: line 6: RUN: command [...]

You trying to build dockerfile, but you running a shell script

srugano, 2024-01-15:
I had the same issue but needed to modify your solution a bit:
      `RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone`

enosjeba, 2024-03-04:
Adding
`ARG DEBIAN_FRONTEND=noninteractive`
worked for me.