microsoft/AirSim #3879: How to run AirSim on headless server

Ryan-Rudes, 2021-07-14:
## Question
### What's your question?
How to run AirSim on headless server?

### Include context on what you are trying to achieve

- I tried to install the Vulcan Driver, but it continues to say that it fails to load
- I started a virtual display, but it does not detect it

#### Context details
Python3.8.8, Unreal 4.25, Ubuntu 18.04.4 LTS

### Logs
```
...
[2021.07.14-14.32.19:864][  0]LogCore: Selected clock_id 1 (CLOCK_MONOTONIC) since it is the fastest support clock without zero deltas.
[2021.07.14-14.32.19:865][  0]LogInit: Unix-specific commandline switches:
[2021.07.14-14.32.19:865][  0]LogInit:  -ansimalloc - use malloc()/free() from libc (useful for tools like valgrind and electric fence)
[2021.07.14-14.32.19:865][  0]LogInit:  -jemalloc - use jemalloc for all memory allocation
[2021.07.14-14.32.19:865][  0]LogInit:  -binnedmalloc - use binned malloc  for all memory allocation
[2021.07.14-14.32.19:865][  0]LogInit:  -filemapcachesize=NUMBER - set the size for case-sensitive file mapping cache
[2021.07.14-14.32.19:865][  0]LogInit:  -useksm - uses kernel same-page mapping (KSM) for mapped memory (OFF)
[2021.07.14-14.32.19:865][  0]LogInit:  -ksmmergeall - marks all mmap'd memory pages suitable for KSM (OFF)
[2021.07.14-14.32.19:865][  0]LogInit:  -preloadmodulesymbols - Loads the main module symbols file into memory (OFF)
[2021.07.14-14.32.19:865][  0]LogInit:  -sigdfl=SIGNAL - Allows a specific signal to be set to its default handler rather then ignoring the signal
[2021.07.14-14.32.19:865][  0]LogInit:  -httpproxy=ADDRESS:PORT - redirects HTTP requests to a proxy (only supported if compiled with libcurl)
[2021.07.14-14.32.19:865][  0]LogInit:  -reuseconn - allow libcurl to reuse HTTP connections (only matters if compiled with libcurl)
[2021.07.14-14.32.19:865][  0]LogInit:  -virtmemkb=NUMBER - sets process virtual memory (address space) limit (overrides VirtualMemoryLimitInKB value from .ini)
[2021.07.14-14.32.19:865][  0]LogInit:  - Physical RAM available (not considering process quota): 377 GB (385606 MB, 394860716 KB, 404337373184 bytes)
[2021.07.14-14.32.19:865][  0]LogInit:  - VirtualMemoryAllocator pools will grow at scale 1.4
[2021.07.14-14.32.19:865][  0]LogInit:  - MemoryRangeDecommit() will be a no-op (re-run with -vmapoolevict to change)
[2021.07.14-14.32.20:436][  0]LogInit: Physics initialised using underlying interface: PhysX
[2021.07.14-14.32.20:509][  0]LogInit: Using OS detected language (en-US).
[2021.07.14-14.32.20:509][  0]LogInit: Using OS detected locale (en-US).
[2021.07.14-14.32.20:510][  0]LogTextLocalizationManager: No specific localization for 'en-US' exists, so the 'en' localization will be used.
[2021.07.14-14.32.20:894][  0]LogSlate: New Slate User Created.  User Index 0, Is Virtual User: 0
[2021.07.14-14.32.20:894][  0]LogSlate: Slate User Registered.  User Index 0, Is Virtual User: 0
[2021.07.14-14.32.21:981][  0]LogRHI: Warning: Could not detect DISPLAY or WAYLAND_DISPLAY environment variables
[2021.07.14-14.32.22:014][  0]LogLinux: Warning: MessageBox: Failed to load Vulkan Driver which is required to run the engine.
The engine no longer fallbacks to OpenGL4 which has been deprecated.: Message: 
[2021.07.14-14.32.22:014][  0]Message dialog closed, result: Cancel, title: Message, text: Failed to load Vulkan Driver which is required to run the engine.
The engine no longer fallbacks to OpenGL4 which has been deprecated.
[2021.07.14-14.32.22:014][  0]LogCore: FUnixPlatformMisc::RequestExit(bForce=true, ReturnCode=1)
[2021.07.14-14.32.22:014][  0]LogCore: FUnixPlatformMisc::RequestExit(1)
```

closed 2021-07-15 by Ryan-Rudes

Tomakko, 2021-09-06:
Hi, did you manage to resolve this issue?

lhjackgod, 2025-05-20:
hello did you resolve this issue?