microsoft/AirSim #3893: To REMOVE a vehicle

hansen-cpu, 2021-07-20:
## What feature are you suggesting?
### Overview:

It would be much better if a function which can remove or delete an existing vehicle.

### Smaller Details: 

With the function:
    def simAddVehicle(self, vehicle_name, vehicle_type, pose, pawn_path = "")
we can create new vehicles. However, there is not a corresponding function which can delete existing vehicles. That would be a trouble when I run the program in a loop, in which I want to repeatedly control a certain number of vehicles, but with different original position.

### Nature of Request:  

- Addition

  - 

## Why would this feature be useful?
Just as what I wrote before.

jonyMarino, 2021-07-20:
Hi @hansen-cpu and welcome! If you need this you can take a look at the PR where simAddVehicle was added (#2390).
Instead of having an addBody() method in AirLib/include/physics/PhysicsWorld.hpp you would have a removeBody using erase_remove() instead of insert().
PRs are welcome!

hansen-cpu, 2021-07-21:
Thank you for your reply! I'll try to figure out that.

closed 2021-07-21 by hansen-cpu