microsoft/AirSim #3894: Gazebo with QGroundControl and PX4 SITL Glitching

nikitabeebe, 2021-07-20:
## Bug report 

- AirSim Version/#commit: 1.5.0
- UE/Unity version: 4.25.4
- autopilot version: PX4 Autopilot gazebo_iris
- OS Version: Ubuntu 18.04

### What's the issue you encountered?
I am trying to run GazeboDrone with a PX4 SITL and QGroundControl. I have connected the PX4 and QGC, configured the settings and compiled a seperate project in order to lauch Gazebo with gcc. However upon running a base environment, the drone as it appears in UE4 is glitching and stuck in hover position no matter what I do with QGC. QGC and gazebo match during flight and I have no compile or glitching on those. But land/takeoff/flyto all render nothing but a static-y drone in UE4. (The rotors also do not move). I am not using any external RC. 

### Settings
AirSim settings
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "PhysicsEngineVersion": "ExternalPhysicsEngine",
  "SimMode": "Multirotor",
  "ClockType": "SteppableClock",
  "Vehicles": {
      "PX4": {
          "VehicleType": "PX4Multirotor",
          "UseSerial": false,
          "LockStep": true,
          "UseTcp": true,
          "TcpPort": 4560,
          "ControlPortLocal": 14540,
          "ControlPortRemote": 14580,
          "Sensors":{
              "Barometer":{
                  "SensorType": 1,
                  "Enabled": true,
                  "PressureFactorSigma": 0.0001825
              }
          },
          "Parameters": {
              "NAV_RCL_ACT": 0,
              "NAV_DLL_ACT": 0,
              "COM_OBL_ACT": 1,
              "LPE_LAT": 47.641468,
              "LPE_LON": -122.140165
          }
      }
  }
}
I have also opened the ports TCD and UDP, but I receive an error upon starting up with AirSim 
Waiting for TCP connection on port 4560, local IP 127.0.0.1
Accepting TCP socket failed, is another instance running?
This error shows on the UE4 gui, not in ./GazeboDrone terminal (which is able to connect without issue)

### How can the issue be reproduced?
1. Follow instructions for setting up PX4 SITL via https://microsoft.github.io/AirSim/px4_sitl/
2. Follow readme of GazeboDrone via https://github.com/microsoft/AirSim/blob/master/GazeboDrone/README.md
3. Launch UE4 editor, followed by gazebo (via PX4 autopilot), and QGC. Then run ./GazeboDrone and attempt to take off via commands in QGC.

### Include full error message in text form
Only actual error that shows is 'Waiting for TCP connection on port 4560, local IP 127.0.0.1
Accepting TCP socket failed, is another instance running?'. Note I am also running ./UE4Editor -opengl4 as it runs smoother on my nuc with ubuntu. 

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2021-07-20:
Hi Nikita! You need to set a simpleflight drone and without steppable clock in your settings.json
So try just deleting the vehicles and clocktype parameters

nikitabeebe, 2021-07-21:
As in the only thing I need in settings.json is this?
`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "PhysicsEngineVersion": "ExternalPhysicsEngine",
  "SimMode": "Multirotor"
}`

Because that gets rid of the TCP error, but it doesn't change the behavior of the drone in UE4.

nikitabeebe, 2021-07-21:
And the glitching starts right as i launch ./GazeboDrone

jonyMarino, 2021-07-21:
> As in the only thing I need in settings.json is this?
>
> `{ "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md", "SettingsVersion": 1.2, "PhysicsEngineVersion": [...]

Yes, you need just that.
In which order are you executing the apps?

nikitabeebe, 2021-07-21:
1. Open UE4 by running ./UE4Editor -opengl4
2. Open my custom (base) project
3.  Open PX4 with the make command for gazebo iris
4. Open QGC
5. Run/Play UE4 (drone pops up without glitching, rotors are spinning)
6. Run ./GazeboDrone
As soon as I run gazebo drone the drone begins to glitch

jonyMarino, 2021-07-21:
What does it do?

nikitabeebe, 2021-07-21:
Let me make a video and upload it

nikitabeebe, 2021-07-21:
Taking off: https://youtu.be/BNDBfRzLaT0
Landing: https://youtu.be/i8kNaL5LZds

jonyMarino, 2021-07-23:
Hi Nikita. The error is in your settings.json. there is no PhysicsEngineVersion parameter It is PhysicsEngine without Version

nikitabeebe, 2021-07-23:
Hmm, unfortunately that didn't change anything. I recompiled with .setup.sh and .build.sh in case, but its the exact same behavior once I launch ./GazeboDrone

jonyMarino, 2021-07-23:
That's weird. From your video it seems that the External physics engine would not be enabled.

nikitabeebe, 2021-07-23:
I'm sorry, I had a typo in the External Physics. So now it does glitch but it simply doesn't move/takeoff... it stays slightly off-ground hovering, but regardless of commands in QGC nothing occurs on UE4.

nikitabeebe, 2021-07-26:
@jonyMarino and you didn't need to modify the .cpp file did you?

jonyMarino, 2021-07-26:
No. You don't need to modify that

jonyMarino, 2021-07-26:
Hi Nikita. I have tried with your settings changing PhysicsEngineVersion by PhysicsEngineName, and I was able to run this sample script without problems:
https://raw.githubusercontent.com/jonyMarino/AirSim/external_dynamics_simulator_with_example/PythonClient/multirotor/move_through_set_vehicle_pose.py
(it doesn't need Gazebo or GazeboDrone)
See if you can make it work.
You are using the binaries, right?

nikitabeebe, 2021-07-26:
Yes that worked. Using the binaries, with the default airsim blocks. I still can't figure why GazeboDrone isn't responding. I can control the drone with a joystick and see the rotors stop moving according to the input, but the physical drone doesn't change location at all. And the outprint from main.cpp with local and global positioning from Gazebo shows appropriate changes in the pose of the drone, but it seems as though the actual 'setVehiclePose' does not succeed to move the drone to that location. Is there something that could be overriding the location or a setting that needs to be changed within Blocks?

nikitabeebe, 2021-07-26:
Off the top of my head though, is there a reason this might work with python and not c++? Thinking of trying to convert main.cpp over to python just for the heck of it...

jonyMarino, 2021-07-26:
Yes, you can try. It shouldn't change anything. Also, you can try re-writing the script I have provided to C++.

nikitabeebe, 2021-07-28:
Resolved by changing line 70 of main.cpp from       
` if (i == msg->pose_size() - 1) {` 
to `if (i==0)`
This uses the correct local pose of the iris drone to change its pose to according to setVehcilePose().

PR #3920 (2021-07-30): Use location instead of GPS for GazeboDrone

Fixes: #3894    

## About

GazeboDrone was using GPS location instead of local frame.

## How Has This Been Tested?

GazeboDrone+PX4+QGroundControl

## Screenshots (if appropriate):
closed 2021-07-28 by nikitabeebe