microsoft/AirSim #3908: When I set BP_FlyingPawn to AirSimGameMode, I got Error：Array index out of bounds: 0 from an array of size 0

mkygogo, 2021-07-25:
## Bug report 

- AirSim Version/#commit:  1.5
- UE/Unity version: ue 4.26
- autopilot version: 
- OS Version: win 10 x64

### What's the issue you encountered?

I need to build a multiplayer level, so I create a new GameMode and inherited from AirSimGameMode, But If I set the Default Pawn Class to BP_FlyingPawn , I got error: Array index out of bounds: 0 from an array of size 0.

### Settings

### How can the issue be reproduced?

1.just set Default Pawn Class to BP_FlyingPawn,  it always break
2.

### Include full error message in text form

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2021-07-29:
Hi @mkygogo! Why are you doing that? You can configure the drones You want to use in the setting.json file

mkygogo, 2021-07-30:
> Hi @mkygogo! Why are you doing that? You can configure the drones You want to use in the setting.json file

I need to controller by manually on different computers , not just flying them just one computer

jonyMarino, 2021-07-30:
You should be able to configure all the drones you want in the settings.json and then have different RPC clients on the different computers.

closed 2022-05-30 by jonyMarino