microsoft/AirSim #3913: Unable to build GazeboDrone

Jaeyoung-Lim, 2021-07-28:
## Bug report 

- AirSim Version/#commit: 
- UE/Unity version: 4.25
- autopilot version: PX4
- OS Version: Ubuntu 18.04

### What's the issue you encountered?

By following the instructions of the [readme](https://github.com/microsoft/AirSim/blob/master/GazeboDrone/README.md), I am unable to build the `GazeboDrone` example

I have read that there may be issues when updating the compiler in `build.sh`, so I have also tried removing `build_release` directory completely however this did not change the error output log. 

### How can the issue be reproduced?

1. Edit [`build.sh`](https://github.com/microsoft/AirSim/blob/6e2512e25d794c9a17c2f45fa94cb441fbc9262b/build.sh#L57) to `gcc-8` and `g++-8`
2. Follow instructions on https://github.com/microsoft/AirSim/blob/master/GazeboDrone/README.md

### Include full error message in text form
```
:~/dev/AirSim/GazeboDrone/build$ make
Scanning dependencies of target GazeboDrone
[ 50%] Building CXX object CMakeFiles/GazeboDrone.dir/src/main.cpp.o
[100%] Linking CXX executable GazeboDrone
/usr/bin/ld: cannot find -lAirLib
/usr/bin/ld: cannot find -lrpc
/usr/bin/ld: cannot find -lMavLinkCom
collect2: error: ld returned 1 exit status
CMakeFiles/GazeboDrone.dir/build.make:216: recipe for target 'GazeboDrone' failed
make[2]: *** [GazeboDrone] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/GazeboDrone.dir/all' failed
make[1]: *** [CMakeFiles/GazeboDrone.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```

**What's better than filing an issue? Filing a pull request :).**

jonyMarino, 2021-07-28:
Hi Jaeyoung! Good to see you again around here.
You need to change line 24 of [this CMakeList.txt](https://www.github.com/microsoft/AirSim/tree/master/GazeboDrone%2FCMakeLists.txt) with the folder where that libraries are located.

Jaeyoung-Lim, 2021-07-28:
@jonyMarino Good to see you again! 

I solved the issue by pointing `LD_LIBRARY_PATH` to `{AIRSIM_ROOT}/build_release` and managed to run the example as shown in https://github.com/microsoft/AirSim/pull/3912 :smile:

closed 2021-07-28 by Jaeyoung-Lim