microsoft/AirSim #3917: msgpackrpc.error.RPCError While Running AirSimNH and hello_drone.py

jakefleisig, 2021-07-30:
## Bug report 
- AirSim Version/#commit: latest
- UE/Unity version: latest
- autopilot version: N/A
- OS Version: Ubuntu 18.04

### What's the issue you encountered?
I am getting the following error when running hello_drone.py while using the AirSimNH environment:
`msgpackrpc.error.RPCError: rpclib: function 'simGetImages' (called with 3 arg(s)) threw an exception. The exception is not derived from std::exception. No further information available.`
I run the AirSimNH executable, and then launch the hello_drone.py script and this is what occurs once it gets to the image calls.

### Settings
I am using the following settings in my setttings.json file:
`{
	"SettingsVersion": 1.2,
	"SimMode":"Multirotor",
	"Vehicles": {
		"Drone1": {
			"VehicleType": "SimpleFlight",
			"AutoCreate": true,
			"Sensors": {
				"Lidar": {
                    			"SensorType": 6,
                    			"Enabled" : true,
                    			"NumberOfChannels": 128,
                    			"RotationsPerSecond": 100000,
                    			"PointsPerSecond": 100000,
					"Range": 50,
                    			"X": 0, "Y": 0, "Z": -1,
                    			"Roll": 0, "Pitch": 0, "Yaw" : 0,
                    			"VerticalFOVUpper": 180,
                    			"VerticalFOVLower": -180,
                    			"HorizontalFOVStart": -180,
                    			"HorizontalFOVEnd": 180,
                    			"DrawDebugPoints": false,
                    			"DataFrame": "SensorLocalFrame"
                		},
				"Gps": {
					"SensorType": 3,
					"Enabled" : true,
					"EphTimeConstant": 0.9,
					"EpvTimeConstant": 0.9,
					"EphInitial": 25,
					"EpvInitial": 25,
					"EphFinal": 0.1,
					"EpvFinal": 0.1,
					"EphMin3d": 3,
					"EphMin2d": 4,
					"UpdateLatency": 0.2,
					"UpdateFrequency": 50,
					"StartupDelay": 1
				},
				"Distance": {
					"SensorType": 5,
					"Enabled": true,
					"MinDistance": 0.1,
					"MaxDistance": 50,
					"X":0, "Y":0, "Z": -1,
					"Yaw": 0, "Pitch": 0, "Roll": 0,
					"DrawDebugPoints": false
				}
		 	}
		}
	}
}`

### How can the issue be reproduced?
1. download unreal, and then airsim.  download the zip file containing AirSimNH from the latest Linux release
2. build/unzip everything and click the AirSimNH executable
3. run hello_drone.py and wait until it gets to the "collect images" prompt.  after that, you should see my error

### Include full error message in text form
Traceback (most recent call last):
  File "hello_drone.py", line 57, in <module>
    airsim.ImageRequest("1", airsim.ImageType.Scene, False, False)])  #scene vision image in uncompressed RGBA array
  File "/home/fleisigj/Beholder/AirSim/PythonClient/airsim/client.py", line 266, in simGetImages
    responses_raw = self.client.call('simGetImages', requests, vehicle_name, external)
  File "/home/fleisigj/.local/lib/python3.6/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/fleisigj/.local/lib/python3.6/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'simGetImages' (called with 3 arg(s)) threw an exception. The exception is not derived from std::exception. No further information available.

rajat2004, 2021-07-30:
You're using the latest master Python package which doesn't work with the 1.5 release binaries for the Camera APIs due to #3320. Try with the 1.5.0 Python package
I think going forward some version info should be added in the docstrings so that people are aware of what has been modified. Version handling inside the code itself might become pretty complicated, but again something to think about

jakefleisig, 2021-07-30:
@rajat2004 which Python package are you referring to? the AirSim python package?

rajat2004, 2021-07-30:
Install the `airsim` package through Pip, and comment out the `import setup_path` from the first line in hello_drone.py. Not adding the setup_path line before importing airsim will use the pip installed one, with setup_path it uses the package from source

jakefleisig, 2021-07-30:
this worked! thank you so much.  i really appreciate the help!

jakefleisig, 2021-07-30:
this worked! thank you very much!

closed 2021-07-30 by jakefleisig