microsoft/AirSim #3921: Collecting rotors data when recording

AbdullahK047, 2021-07-31:
Hi,

I'm interested in collecting the rotors data (i.e, thrusts, speeds, torques) along with the already collected kinematics data when recording. I think the PawnSimApi.cpp file would have to be edited, but I'm not sure how. Can someone help out?

Thanks in advance!
Abdullah

rajat2004, 2021-07-31:
[This thread](https://github.com/microsoft/AirSim/discussions/3624#discussioncomment-651002) might be useful here, just that you would need to use the [getRotorStates()](https://github.com/microsoft/AirSim/blob/880c5541fd4824ee2cd9bb82ca5f611eb1ab236a/AirLib/include/vehicles/multirotor/api/MultirotorApiBase.hpp#L129) method instead of the sensor APIs.
Probably should add this info to the docs also, makes it easier for future

AbdullahK047, 2021-08-02:
> [This thread](https://github.com/microsoft/AirSim/discussions/3624#discussioncomment-651002) might be useful here, just that you would need to use the [...]

Thanks @rajat2004 ! The thread you mentioned was really helpful

closed 2021-08-02 by AbdullahK047