microsoft/AirSim #3923: How to get camera parameters

p00uya, 2021-08-01:
## Adding a new camera to AirSIm like Scene to image processing purposes?
### Overview:

hi everyone, I'm really excited and interested in AIrSIm and Unfamiliar with CV & BP & AirSim 
I want to have a camera to do some CV tasks on it, especial camera I mean! 
on the other hand i need camera to detect edge real time in subwindow and i can procces it , recently i read [this](http://www.huyaoyu.com/technical/2021/04/29/modify-airsim.html)  and i hope to do!

### Smaller Details: 

Now i Need help to find :

- which kind of image has been captured by the Scene camera?(compress? unit8? image as float?)what is the FPS?
- can i use this image to do my task real-time?
- I found some [files](http://www.huyaoyu.com/technical/2021/04/29/modify-airsim.html) that should be modified Now I need help where did I change in "ImageCapture.CPP" file my goal is achieve to image frames that showing on scene part of subwindows 

@shukuizhang @saihv @lovettchris  @sytelus

zimmy87, 2021-08-03:
To answer your questions:

- The image captured by the Scene camera depends upon the parameters set in the ImageRequest object passed to the simGetImages API. By default, ImageRequest.pixels_as_float = False and ImageRequest.compress = False, so pixel data will be stored uncompressed, in an array of uint8's, but that can be configured by setting those two properties.
- This probably depends upon the complexity of your scene. Take a look at the settings for [recording](https://microsoft.github.io/AirSim/settings/#recording). When testing enabling recording in the Blocks environment, I was able to maintain 50-60 FPS, so it's feasible recording can operate in real-time, but this may change for more complex environments and the performance of your system's hardware.
- I'm slightly confused by your last question. Do you mean where in ImageCaptureBase.hpp do you add new image types? This will require updating the ImageType enum defined near the beginning of the file on line ln 19-30. As for obtaining image frames rendered in the subwindows. Look at the default settings for subwindows [here](https://microsoft.github.io/AirSim/settings/#subwindows), each subwindow is configured with a CameraName and ImageType. You'll want to set the camera_name and image_type parameters in the ImageRequest object passed to your simGetImages call to the same values as the subwindow configuration to obtain the same camera images rendered there.

p00uya, 2021-08-04:
hi @zimmy87  Thank you for your reply.
My goal is to see the processed image in the subwindow. I mean, I could process the image captured by the camera the process such as image edge detection or color recognition of objects.
 
for this task first I need to execute image frames then process on it! Now my next step is image frame captured from env. real-time i want to access them firm scene images frame!

zimmy87, 2021-08-11:
For capturing individual image frames and performing processing on them, you'll want to use the [Image APIs](https://microsoft.github.io/AirSim/image_apis/), specifically simGetImage and simGetImages. By specifying an ImageType of airsim.ImageType.Scene (which equals 0), you can grab the currently rendered frame and then perform processing on it.

p00uya, 2021-08-12:
hi @zimmy87 Thank you for your reply.
yeah i test it! it not real time , i do that by addressing file path on the airsim and proccess on image. Now i wanna real Time showing results on the subwindows camera

emersonjr, 2021-08-20:
hey @p00uya ! i've basically trying to accomplish the same thing you're trying as well, did you have any progress on rendering results on the subwindows camera?

p00uya, 2021-08-24:
hey sorry I get confused completely, i don't know where the image frames
saved
if i can access to them is it possible to Transfer to an array and then
process  on them

zimmy87, 2021-09-17:
If doing processing on images obtained from the image APIs is too slow, an alternative option would be to implement whatever processing you want to do as a post process material then add an CaptureComponent to BP_PIPCamera and create an extra ImageType value for selecting the new CaptureComponent. An example of doing an edge detection algorithm is located [here](https://ferkizue.blogspot.com/2018/04/sobel-edge-detection-in-unreal-engine.html)

closed 2023-01-25 by p00uya