microsoft/AirSim #3925: Crash when execute the simEanbleWeather() function

lastbasket, 2021-08-02:
## Bug report 

- AirSim Version/#commit: 1.50 / Fri Jun 25 11:41:14 2021 -0700
- UE/Unity version: 4.25.4
- autopilot version: 
- OS Version: Windows 10 20H2

### What's the issue you encountered?

I run the block in the unreal editor, and it crashes when I call the simEanbleWeather() function in python.
It is fine to run the block without enabling the weather. And I notice that the error says that the engine unable to load the weather assets.

### Settings

### How can the issue be reproduced?

1.
2.

### Include full error message in text form
[2021.08.02-06.56.55:239][403]LogActor: Warning: BP_CameraDirector_C /Game/FlyingCPP/Maps/UEDPIE_0_FlyingExampleMap.FlyingExampleMap:PersistentLevel.CameraDirector has natively added scene component(s), but none of them were set as the actor's RootComponent - picking one arbitrarily
[2021.08.02-06.56.55:315][403]LogLinker: Warning: Asset '../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset' has been saved with empty engine version. The asset will be loaded but may be incompatible.
[2021.08.02-06.56.55:315][403]LogLinker: Warning: Unable to load package (../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset) PackageVersion 522, MaxExpected 518 : LicenseePackageVersion 0, MaxExpected 0.
[2021.08.02-06.56.55:315][403]LogLinker: Warning: Asset '../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset' has been saved with empty engine version. The asset will be loaded but may be incompatible.
[2021.08.02-06.56.55:315][403]LogLinker: Warning: Unable to load package (../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset) PackageVersion 522, MaxExpected 518 : LicenseePackageVersion 0, MaxExpected 0.
[2021.08.02-06.56.55:315][403]LogUObjectGlobals: Warning: Failed to find object 'Class /AirSim/Weather/WeatherFX/WeatherActor.WeatherActor_C'
[2021.08.02-06.56.55:315][403]LogTemp: Warning: Warning, WeatherAPI got invalid weather actor class!
[2021.08.02-06.56.55:316][403]LogLinker: Warning: Asset '../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset' has been saved with empty engine version. The asset will be loaded but may be incompatible.
[2021.08.02-06.56.55:316][403]LogLinker: Warning: Unable to load package (../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset) PackageVersion 522, MaxExpected 518 : LicenseePackageVersion 0, MaxExpected 0.
[2021.08.02-06.56.55:316][403]LogLinker: Warning: Asset '../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset' has been saved with empty engine version. The asset will be loaded but may be incompatible.
[2021.08.02-06.56.55:317][403]LogLinker: Warning: Unable to load package (../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset) PackageVersion 522, MaxExpected 518 : LicenseePackageVersion 0, MaxExpected 0.
[2021.08.02-06.56.55:317][403]LogLinker: Warning: Asset '../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset' has been saved with empty engine version. The asset will be loaded but may be incompatible.
[2021.08.02-06.56.55:317][403]LogLinker: Warning: Unable to load package (../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset) PackageVersion 522, MaxExpected 518 : LicenseePackageVersion 0, MaxExpected 0.
[2021.08.02-06.56.55:317][403]LogLinker: Warning: Asset '../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset' has been saved with empty engine version. The asset will be loaded but may be incompatible.
[2021.08.02-06.56.55:317][403]LogLinker: Warning: Unable to load package (../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset) PackageVersion 522, MaxExpected 518 : LicenseePackageVersion 0, MaxExpected 0.
[2021.08.02-06.56.55:318][403]LogLinker: Warning: Asset '../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset' has been saved with empty engine version. The asset will be loaded but may be incompatible.
[2021.08.02-06.56.55:318][403]LogLinker: Warning: Unable to load package (../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset) PackageVersion 522, MaxExpected 518 : LicenseePackageVersion 0, MaxExpected 0.
[2021.08.02-06.56.55:318][403]LogLinker: Warning: Asset '../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset' has been saved with empty engine version. The asset will be loaded but may be incompatible.
[2021.08.02-06.56.55:318][403]LogLinker: Warning: Unable to load package (../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset) PackageVersion 522, MaxExpected 518 : LicenseePackageVersion 0, MaxExpected 0.
[2021.08.02-06.56.55:318][403]LogLinker: Warning: Asset '../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset' has been saved with empty engine version. The asset will be loaded but may be incompatible.
[2021.08.02-06.56.55:318][403]LogLinker: Warning: Unable to load package (../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset) PackageVersion 522, MaxExpected 518 : LicenseePackageVersion 0, MaxExpected 0.
[2021.08.02-06.56.55:319][403]LogLinker: Warning: Asset '../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset' has been saved with empty engine version. The asset will be loaded but may be incompatible.
[2021.08.02-06.56.55:319][403]LogLinker: Warning: Unable to load package (../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset) PackageVersion 522, MaxExpected 518 : LicenseePackageVersion 0, MaxExpected 0.
[2021.08.02-06.56.55:320][403]LogLinker: Warning: Asset '../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset' has been saved with empty engine version. The asset will be loaded but may be incompatible.
[2021.08.02-06.56.55:320][403]LogLinker: Warning: Unable to load package (../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset) PackageVersion 522, MaxExpected 518 : LicenseePackageVersion 0, MaxExpected 0.
[2021.08.02-06.56.55:320][403]LogLinker: Warning: Asset '../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset' has been saved with empty engine version. The asset will be loaded but may be incompatible.
[2021.08.02-06.56.55:320][403]LogLinker: Warning: Unable to load package (../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset) PackageVersion 522, MaxExpected 518 : LicenseePackageVersion 0, MaxExpected 0.
[2021.08.02-06.56.55:320][403]LogLinker: Warning: Asset '../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset' has been saved with empty engine version. The asset will be loaded but may be incompatible.
[2021.08.02-06.56.55:320][403]LogLinker: Warning: Unable to load package (../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset) PackageVersion 522, MaxExpected 518 : LicenseePackageVersion 0, MaxExpected 0.
[2021.08.02-06.56.55:320][403]LogLinker: Warning: Asset '../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset' has been saved with empty engine version. The asset will be loaded but may be incompatible.
[2021.08.02-06.56.55:320][403]LogLinker: Warning: Unable to load package (../../../../../Code/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/Weather/WeatherFX/WeatherActor.uasset) PackageVersion 522, MaxExpected 518 : LicenseePackageVersion 0, MaxExpected 0.
[2021.08.02-06.56.55:330][403]LogTemp: SimpleFlight
[2021.08.02-06.56.55:333][403]PIE: Server logged in

And here is the link for the full log:
[https://drive.google.com/file/d/1iFEa4DXYpat1WLv1r5H_Co8TrVpS_vQw/view?usp=sharing](url)

jonyMarino, 2021-08-03:
Hi @lastbasket! That was a known issue in the latest version. Can you try cloning the repository, please?

lastbasket, 2021-08-11:
Thanks, I pull the latest and it is fine now.

closed 2021-08-11 by jonyMarino