microsoft/AirSim #3926: Connect real drone to unreal airsim

eldewany, 2021-08-02:
## Question
### What's your question?
I want to connect the real drone ( Pixhawk with jetson nano) to the airsim simulation so that I can mimic the real drone on the sim. I managed to run drone server and everything but I can't connect the unreal simulation to the drone, it either shows me a false connection and nothing happens when the drone works, so what can I do?.

#### Context details
Airsim: v1.5
Pixhawk: PX4 v1.12 / v 1.11.3
Unreal: uploaded official builds
settings:

{
    "SettingsVersion": 1.5,
    "SimMode": "Multirotor",
    "ClockType": "SteppableClock",
    "Vehicles": {
        "PX4": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "LockStep": true,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlIp": "remote",
            "ControlPortLocal": 14580,
            "ControlPortRemote": 14585,
            "LocalHostIp": "0.0.0.0",
            "QgcHostIp": "192.168.0.36",
            "QgcPort": 14550,
            "Sensors":{
                "Barometer":{
                    "SensorType": 1,
                    "Enabled": true,
                    "PressureFactorSigma": 0.0001825
                }
            },
            "params": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "COM_OBL_ACT": 1
            }
        }
    }
}

zimmy87, 2021-08-03:
It looks like you're using settings for PX4 SITL instead of HITL. Please take a look at the documentation [here](https://microsoft.github.io/AirSim/px4_setup/). The "UseSerial" setting should be set to "true".

eldewany, 2021-08-04:
> It looks like you're using settings for PX4 SITL instead of HITL. Please take a look at the documentation [here](https://microsoft.github.io/AirSim/px4_setup/). The "UseSerial" setting should be set to "true".

It's not really my case of use because I'm going to run them remotely through udp or tcp connection.