microsoft/AirSim #3927: RPC Error getting Imu Data

nikitabeebe, 2021-08-02:
## Question
Trying to run Hello Drone in python client on my Ubuntu and I receive an error when getting the IMU data. Is this an issue with a dependency I'm missing or do I need additional configurations? 

### Include context on what you are trying to achieve
Running Hello_Drone.py in the binary Blocks environment with no customization to settings.json. Once I launch Hello_drone is shows 'connected' but then errors out on the line, imu_data = client.getImuData().

#### Context details
Ubuntu 18.04, UE 4.25.4, latest blocks and airsim, python 3.7.6, GCC 7.3.0
Default settings.json
`Traceback (most recent call last):
  File "hello_drone.py", line 19, in <module>
    imu_data = client.getImuData()
  File "/home/summer-dev/repos/airsim/PythonClient/airsim/client.py", line 679, in getImuData
    return ImuData.from_msgpack(self.client.call('getImuData', imu_name, vehicle_name))
  File "/home/summer-dev/anaconda3/lib/python3.7/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/summer-dev/anaconda3/lib/python3.7/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'getImuData' (called with 2 arg(s)) threw an exception. The exception is not derived from std::exception. No further information available.
`

### Include details of what you already did to find answers
I get the Multirotor state vector without issue. I have tried running this on my own project and still get the same error. This occurs when I use 'conda activate' as well as the direct command 'python hello_drone.py'. The only thing that changes on the block UE4 GUI is it shows API connection was successful.

xxEoD2242, 2021-08-02:
Is the IMU listed in your `settings.json`? Also, are you using SimpleFlight or PX4?

xxEoD2242, 2021-08-02:
You might try adding the sensor name to the call if you add it to your settings.json. For PX4, the guide is wrong and the regular sensor suite are actually lower case, so `imu` instead of `Imu`.

`imu_data = client.getImuData(imu_name = "", vehicle_name = "")`

nikitabeebe, 2021-08-03:
I'm not using PX4, just SimpleFlight

xxEoD2242, 2021-08-03:
Hm...Are you using the setup_path file or binaries for adding AirSim to Python?

nikitabeebe, 2021-08-03:
I've done both. One using pip install airsim and commenting out the setup_path, as well as using the repo clone and a conda environment. Slightly different line errors but the same resulting rpc error with imu data.

xxEoD2242, 2021-08-03:
> I'm not using PX4, just SimpleFlight.
> But if I edit the settings.json to be the following, I still get the same error:
> `{ "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md", "SettingsVersion": 1.2, "DefaultSensors": { "Imu": { "SensorType": 1, "Enabled": true, "AngularRandomWalk": 0.3, "GyroBiasStabilityTau": 500, "GyroBiasStability": 4.6, "VelocityRandomWalk": 0.24, "AccelBiasStabilityTau": 800, "AccelBiasStability": 36 } } }`

Hm...I just checked the docs and it appears you need to add a vehicle, name it and then add in the sensors directly.

```
"Sensors": {
            "Barometer":{
                "SensorType": 1,
                "Enabled": true,
                "PressureFactorSigma": 0.0001825
            },
            "MyLidar1": {
                "SensorType": 6,
                "Enabled" : true,
                "NumberOfChannels": 16,
                "PointsPerSecond": 10000,
                "X": 0, "Y": 0, "Z": -1,
                "DrawDebugPoints": true
            },
            "MyLidar2": {
                "SensorType": 6,
                "Enabled" : true,
                "NumberOfChannels": 4,
                "PointsPerSecond": 10000,
                "X": 0, "Y": 0, "Z": -1,
                "DrawDebugPoints": true
            }
        }
```

jonyMarino, 2021-08-03:
Nikita, that is a bug that was present in the latest binaries (v1.5). Are you using one of them? The latest commit on master should work.

nikitabeebe, 2021-08-03:
Yes I was using the v1.5 binary. I'll try re-cloning from the master branch and running that Blocks environment

nikitabeebe, 2021-08-03:
Yup, using the latest master commit fixed the issue. Thank you!

closed 2021-08-03 by nikitabeebe